FlexBE/flexbe_app #1: Can we load a particular configuration file via command line / roslaunch?

dcconner, 2017-02-28:
We have option to save and load configuration files, but can we load a particular one (mainly to set default package path)

Also, can we write/read config with new lines after each argument?

pschillinger, 2017-02-28:
This is not possible right now, but I also don't see any problems with 
adding this functionality.

Are you referring to new lines in the generated config file? This is 
done by the Javascript JSON package, not sure if new lines are supported 
out of the box.

dcconner, 2017-03-01:
If you want to give me a pointer to the correct spot to change, I can take a look.

Yes.  The current generated config is all on one line when I view in my editor. 
I added new lines when I was making a change, and it seemed to read in OK.

pschillinger, 2017-03-01:
Right now there is only one arg, it is processed by the [main function](
https://github.com/FlexBE/flexbe_app/blob/master/src/main.js#L18). But it
should be ok to add the new one there as well, I will refactor it when it
becomes more.

A config is imported [here](
https://github.com/FlexBE/flexbe_app/blob/master/src/ui/ui_settings.js#L311),
but the interface includes that the file dialog opens and the resulting
entry (not a path) is processed. I would suggest you extract the loading
part from there and let the class offer another public function to import
settings by accepting a file path. [readFile](
https://github.com/FlexBE/flexbe_app/blob/master/src/io/io_filesystem.js#L42)
gives you the file content.

dcconner, 2017-03-01:
For later reference:
ui_settings.js#L335
var content = JSON.stringify(config, null, '\n');