FlexBE/flexbe_app #2: Error with "status label"

dcconner, 2017-03-03:
Running the example_behavior  with App version 2.0.1
```
chrome-extension://bkpceecdjoihccpjecmbafbjifpefbad/src/ui/ui_runtimecontrol.js:113
		status_label = R.text(R.width / 2, R.height - 30, text)
		                ^

TypeError: Cannot read property 'text' of undefined

    at drawStatusLabel (chrome-extension://bkpceecdjoihccpjecmbafbjifpefbad/src/ui/ui_runtimecontrol.js:113:19)
    at Object.UI.RuntimeControl.displayOutcomeRequest (chrome-extension://bkpceecdjoihccpjecmbafbjifpefbad/src/ui/ui_runtimecontrol.js:721:4)
    at outcome_request_callback (chrome-extension://bkpceecdjoihccpjecmbafbjifpefbad/src/rc/rc_pubsub.js:44:21)
    at process.nextTick (chrome-extension://bkpceecdjoihccpjecmbafbjifpefbad/src/ros/ros_subscriber.js:50:59)

```
In method src/ui/ui_runtimecontrol.js  at L111
```
	var drawStatusLabel = function(text) {
		if (status_label != undefined) status_label.remove();
		status_label = R.text(R.width / 2, R.height - 30, text)
			.attr({'font-size': 16, 'fill': 'gray'});
	}

```
There is no protection against R being undefined and needing to initialize

dcconner, 2017-03-03:
I should not this does not always happen, but I have seen this error multiple times on different statemachines

pschillinger, 2017-03-03:
Looks like there is a race condition causing this error if an outcome request is processed before the state display is actually drawn. I will check.

commit 4301720c (2017-03-23): Only update drawing on outcome request if available (fixes #2)
closed 2017-04-23 by pschillinger