FlexBE/flexbe_app #4: Problem creating new behaviours

capox95, 2017-04-02:
Hi, I have some problems when I try to create new behaviors. 
Here the error message:
[link](https://drive.google.com/file/d/0B0k2QgYGT-HKbmhoOXQtellneEU/view)

If I use the default behaviors available I have no problem at all!
(but if I load them, change the name and then I try to save, here we have the error!)

Obviously I am not able to see my new behavior on the "select behaviors" column! 

Behaviors Packages: "flexbe_behaviors"
State Packages: "flexbe_states"

I think that could be a problem with my configuration or maybe with the app because with the Chrome_App everything was fine.

Thanks,
Alessio

dcconner, 2017-04-06:
We are seeing this as well.  As a work around if we "touch new_behavior_sm.py"  and create an empty file first, then it seems to work OK afterwards.

dcconner, 2017-04-06:
Sometimes we don't get any output to the terminal log screen, but sometimes we get "Code Generation completed" and "Manifest generation completed", but then error
"file new_behavior_sm.py not found in specified folder"  (where new_behavior was our test name) and not files are actually created (confirmed in master and develop branches)

pschillinger, 2017-04-06:
There was a bug preventing to create a new behavior if the onboard engine was running and the app was connected to it.

David, the error which shows no terminal output has a different reason, but should be fixed now as well.

Please let me know if there is still another problem with creating behaviors I am not aware of.

TGiles, 2017-04-10:
We tested with latest change, and it saves behavior but when we try to run
```
[ERROR] [WallTime: 1491867753.906567] bad callback: <bound method BehaviorLauncher._callback of <flexbe_widget.behavior_launcher.BehaviorLauncher object at 0x7f9d2345c990>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "/home/cpsc495_team1/Desktop/cpsc495/src/flexbe_behavior_engine/flexbe_widget/src/flexbe_widget/behavior_launcher.py", line 38, in _callback
    be_id, behavior = self._behavior_lib.find_behavior(msg.behavior_name)
TypeError: 'NoneType' object is not iterable
```

If we exit FlexBE then restart, then the new behavior is waiting on us and works.

pschillinger, 2017-04-18:
Ok, makes sense. The behavior library of the launcher only updates once when being started. I will fix this.

pschillinger, 2017-04-18:
Updating the flexbe_behavior_engine repo should fix the above error, at least it works for me now.

matteopantano, 2017-04-20:
I have tried to update the two packages, switching to feature/flexbe_app in flexbe_behavior_engine and switching to develop in flexbe_app but when I try to save my behavior nothing happens, terminal does not give me any response. Furthermore if I try to push the Start Execution in Runtime Control nothing happens.
Am I missing something?

pschillinger, 2017-04-23:
It should not be necessary to disconnect ROS. As I wrote above:
> There was a bug preventing to create a new behavior if the onboard engine was running and the app was connected to it.

Consequently, it was a workaround to disconnect ROS. But this should not be required anymore. Since I am not able to reproduce what you describe, can you give me a few more details regarding your workspace setup and the behavior which fails to save such that I can try to reproduce?

matteopantano, 2017-04-24:
Well behavior is quite simple, I just have one state regarding opening and closing of the gripper (I am using this one in another behavior and it works). When i save I have no errors however if I try to push the button start execution nothing happens. As I said before if I disconnect the master the behavior saves and I can access to it the next time I will open the app, on the other hand if I save it w/out disconnecting i will not find it in the list. 

The behaviors I am using are in folder youbot_behaviors antecedently downloaded from the github repository [youbot_behaviors](https://github.com/FlexBE/youbot_behaviors).

And [here](https://drive.google.com/open?id=0B55MEtAHy9REanJXNGppYjMxVkU) some pic of the setup.

pschillinger, 2017-04-24:
Thank you! One thing that looks surprising to me is that on the flexbe_conf screenshot it looks like youbot_flexbe_states (which contains the gripper state) is not detected as state package. The youbot_behaviors repo has not been converted to the new flexbe_app format, did you do this yourself? Maybe something related to this is causing the issue.

I tested it and pushed a new [feature/flexbe_app](https://github.com/FlexBE/youbot_behaviors/tree/feature/flexbe_app) branch on youbot_behaviors. Can you try again on this branch and the current version of flexbe_behavior_engine and flexbe_app? In the App, the version indicator (top right corner) should be green.

matteopantano, 2017-05-06:
Hi Philip,
Until today it worked. then as long the banner for updating was coming up I updated it and now it does not start any more.

`Traceback (most recent call last):
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_widget/bin/be_launcher", line 35, in <module>
    launcher = BehaviorLauncher()
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_widget/src/flexbe_widget/behavior_launcher.py", line 32, in __init__
    self._behavior_lib = BehaviorLibrary()
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 23, in __init__
    self.parse_packages()
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 34, in parse_packages
    self._add_behavior_manifests(self._rp.get_path(pkg), pkg)
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 51, in _add_behavior_manifests
    self._add_behavior_manifests(entry_path, pkg)
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 53, in _add_behavior_manifests
    m = ET.parse(entry_path).getroot()
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1182, in parse
    tree.parse(source, parser)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 656, in parse
    parser.feed(data)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1642, in feed
    self._raiseerror(v)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1506, in _raiseerror
    raise err
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 7, column 27
Traceback (most recent call last):
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_onboard/bin/start_behavior", line 13, in <module>
    VigirBeOnboard()
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_onboard/src/flexbe_onboard/flexbe_onboard.py", line 61, in __init__
    self._behavior_lib = BehaviorLibrary()
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 23, in __init__
    self.parse_packages()
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 34, in parse_packages
    self._add_behavior_manifests(self._rp.get_path(pkg), pkg)
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 51, in _add_behavior_manifests
    self._add_behavior_manifests(entry_path, pkg)
  File "/home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 53, in _add_behavior_manifests
    m = ET.parse(entry_path).getroot()
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1182, in parse
    tree.parse(source, parser)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 656, in parse
    parser.feed(data)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1642, in feed
    self._raiseerror(v)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1506, in _raiseerror
    raise err
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 7, column 27
[behavior_launcher-3] process has died [pid 30304, exit code 1, cmd /home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_widget/bin/be_launcher __name:=behavior_launcher __log:=/home/matteo/.ros/log/aa9a34fa-3298-11e7-adc3-54271e0d97df/behavior_launcher-3.log].
log file: /home/matteo/.ros/log/aa9a34fa-3298-11e7-adc3-54271e0d97df/behavior_launcher-3*.log
[behavior-4] process has died [pid 30305, exit code 1, cmd /home/matteo/flexbe_ws/src/flexbe_behavior_engine/flexbe_onboard/bin/start_behavior __name:=behavior __log:=/home/matteo/.ros/log/aa9a34fa-3298-11e7-adc3-54271e0d97df/behavior-4.log].
log file: /home/matteo/.ros/log/aa9a34fa-3298-11e7-adc3-54271e0d97df/behavior-4*.log
`

I tried to install it to a new workspace but the error is still the same.

matteopantano, 2017-05-06:
Except from initial errors now seems to work, probably one thing missing is the possibility to delete a behaviour.

astumpf, 2018-01-26:
Hey,
we are using FlexBE App in our lecture. Many students (myself included) have the described issue above. It works usually, but sometimes FlexBE App refuses to save. There is neither an error message nor any strange visible behavior. The app pretends to have saved but indeed didn't.
Any guess?

pschillinger, 2018-01-28:
Which of the issues discussed above?

> It works usually, but sometimes FlexBE App refuses to save.

Have you observed any difference between the cases where it worked and where not? For example, if a ROS connection is active, the kind of changes before saving, etc?

It is hard to make a guess without any context...

astumpf, 2018-01-28:
In the meantime we figured out the issue: You have to set the "Package" at least once, although it does already have a selection. It seems as the app didn't initialize the internal value from the drop down menu initially.

pschillinger, 2018-03-04:
@astumpf I started to look into this, but cannot reproduce it. Can you confirm you were using a version which included the commit https://github.com/FlexBE/flexbe_app/commit/68d8257be76f991ac6bb46121bc032a1dc4f2110?

closed 2018-06-30 by pschillinger

hbaqueiro, 2020-03-10:
@astumpf I'm facing the same problem. How could you manage to set the Package? My package dropdown menu is empty, so I can't set it.