FlexBE/flexbe_app #10: Improvement: Add option to reload/refresh state modules from Statemachine Editor

josephcoombe, 2017-12-06:
If I am prototyping a state machine with some custom state modules, I often realize that I want to modify a state class in some way _(e.g. add/remove parameters, add/remove outcomes)_.

I can edit the module file and make my desired changes; however, those changes are not reflected in the Statemachine Editor GUI.

Instead I have to save my behavior (not always possible), and close/open the FlexBE app.

---
It might be beneficial to have a 'reload/refresh' button in the Statemachine Editor to update the list of available states and update the state block within the Editor.

Then the workflow would be: Epiphany -> edit module file -> click  'refresh/reload' in Editor -> ...

pschillinger, 2017-12-07:
I agree, this would be useful. In fact, I have a feature on my to-do list
which adds an automatic reload whenever you save any state file in your
editor. I did not prioritize it that high because it is rather a
convenience feature, but I consider this issue as an upvote and might in
fact have some time during the weeks after Christmas to implement it.

pschillinger, 2017-12-17:
I implemented a first version of the updating feature. Does this improve your workflow as desired?

Please let me know any feedback about how the feature works for you or if you found bugs there in the corresponding PR #13. Thank you!

josephcoombe, 2017-12-18:
Checked it out. PR #13 does exactly what I wanted and improves my workflow using FlexBE. Thank you!

I only had time to do a quick surface-level run-through, but if I encounter any bugs, I will update PR #13  if it hasn't been merged in yet!

closed 2017-12-18 by josephcoombe