FlexBE/flexbe_app #11: UI: If there are multiple state outcome nets between two states, then their tags will overlap.

josephcoombe, 2017-12-06:
Suppose you have 2 states A and B, each with two outcomes 'x' and 'y'

If you connect state A's outcome 'x' to state B, and  
state B's outcome 'y' to state A, then  
the state outcome nets between the two states (and their tags) will overlap,  
meaning that you will only see one line and either 'x' or 'y'.

---
Perhaps, nets between 2 states should be routed differently (based on origin). Then you'd see 2 distinct lines between state A and state B. One line for the transition from A to B and one for the transition from B to A.

In the case of multiple transitions in one directions, I think it would be best to only render one line and simply display separate tags for each transitions _(e.g. If both 'x' and 'y' from state A are connected to state B, then only one line with separate 'x' and 'y' bubbles would be displayed)_

pschillinger, 2017-12-07:
Yes, good point.

As it is implemented right now, a transition only knows the states it
connects when being drawn, but does not consider any other transition. So
it is in fact not trivial to make the visual appearance of a transition
dependent on others, especially since the drawing order is random.

I can think a bit more about what makes most sense here. If you have any
further ideas, let me know. But I definitely agree with the general problem.

AndyIII, 2018-03-07:
``