FlexBE/flexbe_app #12: Statemachine Editor: Connecting state outcomes becomes progressively slower based on the number of existing state outcome nets

josephcoombe, 2017-12-06:
Was using the Statemachine Editor to edit a sm with 10 states and ~25 state outcome transitions total.

Once I passed about 20 state outcome transitions (i.e. I had made 20 connections between states), I noticed that making new connections became progressively slower (and laggier). 

I'd click on the state outcome name in the origin state and the screen would update slowly (and freeze for fractions of a second) while I tried to drag the net to the destination state. 

This lagginess only manifests itself when making state transition connections and appears to get worse  as the number of state transitions connections made increases.

(I'm working with 32 MB of RAM with a good processor and graphics card). Thanks!

pschillinger, 2017-12-07:
Thank you for letting me know. I am a bit surprised that it is that slow
already for relatively few states, but yes, the visual update when
connecting a transition is implemented rather ineffeciently if I recall
correctly.

I will make sure to improve efficiency along the next updates.

pschillinger, 2017-12-15:
Should be much smoother now, not dependent on the number of transitions anymore, and also update a transition while hovering over states. Hope this improves the performance for you.

josephcoombe, 2017-12-18:
Yep, loaded the same Behavior with the changes in PR #13 and was able to edit the transition net lists without the previous lag. Thank you!

commit 6c963b9c (2017-12-15): Fix #12: Improved responsiveness of connecting transitions
closed 2017-12-18 by josephcoombe