FlexBE/flexbe_app #14: State file name overwritten in Behavior source code when saving

RhysMcK, 2018-04-09:
I have pulled the latest development repo in order to utilise this new feature. However I seem to have come across a bug which prevents me from running the behavior. Every time I save the behavior the file name of the State ( in this case- jp_picking_main.py)  changes in the behavior source code. As seen below on 2nd last line.
## Before saving:
```python
#!/usr/bin/env python
# -*- coding: utf-8 -*-
###########################################################
#               WARNING: Generated code!                  #
#              **************************                 #
# Manual changes may get lost if file is generated again. #
# Only code inside the [MANUAL] tags will be kept.        #
###########################################################

from flexbe_core import Behavior, Autonomy, OperatableStateMachine, ConcurrencyContainer, PriorityContainer, Logger
from kr3_flexbe_states.jp_picking_main import JP_FindAndPick
from flexbe_states.log_state import LogState
```
## After saving:
```python
#!/usr/bin/env python
# -*- coding: utf-8 -*-
###########################################################
#               WARNING: Generated code!                  #
#              **************************                 #
# Manual changes may get lost if file is generated again. #
# Only code inside the [MANUAL] tags will be kept.        #
###########################################################

from flexbe_core import Behavior, Autonomy, OperatableStateMachine, ConcurrencyContainer, PriorityContainer, Logger
from kr3_flexbe_states..goutputstream-C9ESGZ import JP_FindAndPick
from flexbe_states.log_state import LogState
```

Any thoughts? cheers!

RhysMcK, 2018-04-09:
After doing some further testing this seems to just be a bug with flexbe_full.launch.  I am able to save the behavior and also make online edits to the state source file when using flexbe_ocs.launch without any issues. I can then open flexbe_full.launch to run the behavior. However, if i make any changes which require me to resave the behavior when in flexbe_full, the file name of the state will be overwritten as detailed above.

pschillinger, 2018-04-10:
Thank you for reporting! I will look at this as soon as possible.

>

pschillinger, 2018-04-10:
I was able to reproduce the issue and it looks like the following is going on:

When using some editor commands, for example "subl -n $FILE" for sublime, all works fine.

However, when using some other editor commands, including the default one "gedit --new-window $FILE +$LINE", the described issue occurs. The reason is that gedit creates a temporary file called ".goutputstream-xxxxxx", changes this file, and finally renames it to the target file again. As a consequence, the "change" event assumes a wrong file name.

pschillinger, 2018-04-10:
I changed the condition to determine whether a state definition update should be performed. It works for me now.

Can you confirm?

RhysMcK, 2018-04-10:
Thanks @pschillinger , that did it. Works a treat!

commit d2f3de72 (2018-04-10): Fix #14: Update state definition only for python files but any event type
closed 2018-06-30 by pschillinger