FlexBE/flexbe_app #15: Problem with states which have the same type

GabrielHuettenberger, 2018-05-15:
When having two state types with identical names in different state packages, the App only shows one of the two types twice when using the Add State feature. Which type is shown twice seems to change randomly on App restart.
[image]
[image]

pschillinger, 2018-06-30:
Thanks for reporting, I will look into it.

Independent of this, I would suggest to give rather specific names to states and make sure that states are indeed different in the sense that they are more than just two different hardcoded parametrizations of the same functionality. But of course, the issue you reported needs to be fixed.

pschillinger, 2018-11-30:
This problem should be fixed now by the above referenced commit.

In short:

* Duplicate state class names are now correctly handled within the app.
* For generated code, duplicate state classes are imported with according aliases. This enables to correctly load the respective state class when loading a behavior in the app.
* There is a new entry in the settings to do the above explicitly for all state classes even if unique.

commit e4b90f3a (2018-11-30): Fix #15: Correctly handle duplicate state class definitions
closed 2018-12-01 by pschillinger