FlexBE/flexbe_app #17: Multiple instances of Flexbe

RhysMcK, 2018-06-18:
Hey @pschillinger ,

I am trying to control two difference robots by launching flexbe_app (`roslaunch flexbe_app flexbe_full.launch`) in two different namespaces, however the Flexbe_app seems to become unresponsive when i launch the 2nd instance. Is flexbe_app currently not support to run multiple instances, if so, is there a way to get this to work?

Thanks Phil!

pschillinger, 2018-06-30:
Unfortunately, this results from a limitation of the nwjs framework. In general, only one instance of an app can be running (i.e., one process). It is possible to open multiple additional windows in the same process and thus, pretend multi-instance apps, but this does not work well with ROS.

The main problem is that environment variables like the ROS master URI and also the namespace in which the flexbe_app should be running is given by the respective process and naturally, running a second app in the same namespace does not make much sense.

There might be a way around it, but it is non-trivial and I did not yet look too much into it. If you have any ideas how to address this, I am happy to discuss and will also look again myself if I find a good approach.

manox, 2019-03-07:
Hi, as I ran into the same issue (want to control 2 robots with different ROS-Masters from one PC with 2 FlexBE apps), I wanted to ask if there is already a solution for that. Thanks!

hectorherrero, 2021-02-12:
Hi, are there any updates on this? Our scenario is a bit different, we have one robot running onboard behaviors and want to control/supervise from different PCs. The FlexBE Apps are excluding between them, when a new FlexBE App is lauched blocks the previous one.
Thanks!

tropappar, 2021-11-05:
This pull request allows to run the behavior engine in a sub-namespace, hence allowing multiple instances to run simultaneously.
https://github.com/team-vigir/flexbe_behavior_engine/pull/155

dcconner, 2023-04-28:
See https://github.com/team-vigir/flexbe_behavior_engine/pull/160 

Noting this discussion for ROS 2 changes