FlexBE/flexbe_app #27: Load default behaviour when launching flexbe_full.launch?

Nachmr, 2018-11-29:
I just finished the development of a state machine using FlexBE, and now I have a launch file that starts all the nodes my robot requires and set them ready to operate. 
Is it possible to either add a specific argument or change some parameters in FlexBE configuration, so my behaviour loads automatically instead of FlexBE starting with no behaviour loaded and for me to me have to click in load `behaviour > my_behavior`? Maybe I should use another way of launch FlexBE rather than flexbe_full launch file?

pschillinger, 2018-11-30:
You can use one of the methods described in the tutorial [Running Behaviors Without Operator](http://wiki.ros.org/flexbe/Tutorials/Running%20Behaviors%20Without%20Operator).

There is no method yet to load a behavior in the app directly. But if you use the above method, you don't need the app to start a behavior. If you want to monitor execution, you can still attach the app afterwards.

Is this sufficient or would you request some additional functionality?

Nachmr, 2018-12-10:
@pschillinger Yes it is! Attaching the app afterwards runs smooth. Thanks!

closed 2018-12-10 by Nachmr