FlexBE/flexbe_app #29: view code does not work

mysablehats, 2018-12-14:
Hi, 

I am not sure if this is a bug or not. Basically, opening source code for a state using the button view source, errors out with the message:
Unable to open state in editor: TypeError: Cannot read property 'getFilePath' of undefined

I've described the bug maybe a bit better here ( https://answers.ros.org/question/310698/flexbe-wont-show-source-code/ ), but I thought it was worth opening an issue as well. 

I believe that this is a problem in ui_panels_stateproperties.js l:676, where WS.Statelib.getFromLib(state_class) is returning nonsense. 

I get from that that maybe my statelib is not configured properly (?) and I wonder what I did wrong. 

Thank you very much from the help!

pschillinger, 2018-12-18:
Thank you for reporting the bug! You did nothing wrong, the error was due to a recent change and I fixed it now. Sorry for the inconvenience!

closed 2018-12-18 by pschillinger

mysablehats, 2018-12-19:
Changes work as expected. Thanks!

commit d5c5d847 (2018-12-18): Fix #29: Use correct statelib call to open source code