FlexBE/flexbe_app #32: can't find any state types in type selection

hamma95, 2019-02-17:
I am trying to complete the "hello world" tutorial from the ros flexbe wiki.i followed the tutorials closely but cannot seem to find any state types even though i installed all the state packages mentionned in the tutorial and they are detected in the flexbe app.
here is a screenshot:
[image]

pschillinger, 2019-02-17:
Do you see the entry flexbe_states when you go to the Configuration panel
in the app?

If not, can you run "roscd flexbe_states" in the same terminal where you
launched the app before? Does it work?

If yes, can you check whether the package.xml of flexbe_states contains the
export statement for flexbe_states?

Not sure right now what the problem is because some states are apparently
found, but not the standard ones which come with FlexBE.

hamma95, 2019-02-17:
@pschillinger thank you for your quick reply.this is what my configuration panel looks like:
[image]

hamma95, 2019-02-17:
i solved it. i don't know what the issue was but deleting my workspace and reinstalling everything made it work.i already did try this before posting here,but i forgot to remove the old shortcut and create a new one.what gave me the idea was when i read your reply and launched the app with roslaunch i found the states(before uninstalling everything i didn't even with roslaunch) .Thank you again for the quick reply.

closed 2019-02-18 by hamma95