FlexBE/flexbe_app #35: Not able to install FlexBE app

askkvn, 2019-04-27:
Singularity hrwros-05.simg:~/hrwros_ws/src> **rosrun flexbe_app run_app**
/home/kevin/hrwros_ws/src/flexbe_app/nwjs/nw: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory

pschillinger, 2019-04-27:
It looks like the system is missing a dependency, can you try whether the following installation fixes it:

    sudo apt install libnss3

askkvn, 2019-04-28:
I have added libnss3 and other dependencies. But then getting these error message 

```
Singularity hrwros-05.simg:~> rosrun flexbe_app run_app
Gtk-Message: Failed to load module "canberra-gtk-module"
Gtk-Message: Failed to load module "canberra-gtk-module"
[9855:9855:0428/094526.455387:ERROR:browser_dm_token_storage_linux.cc(93)] Error: /etc/machine-id contains 32 characters (32 were expected).
[9855:9860:0428/094526.535889:ERROR:bus.cc(394)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.
[9855:9884:0428/094526.838529:ERROR:bus.cc(394)] Failed to connect to the bus: Failed to connect to socket /run/user/1000/bus: No such file or directory

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.

(nw:9855): dconf-CRITICAL **: unable to create directory '/run/user/1000/dconf': Permission denied.  dconf will not work properly.
[9855:9873:0428/094529.348185:ERROR:nss_util.cc(674)] Error initializing NSS with a persistent database (sql:/home/kevin/.pki/nssdb): NSS error code: -8023
[9855:9873:0428/094529.348310:ERROR:nss_util.cc(154)] Error initializing NSS without a persistent database: NSS error code: -8023
[9855:9873:0428/094529.348326:FATAL:nss_util.cc(156)] nss_error=-8023, os_error=0
#0 0x7f196b041c8c <unknown>

/home/kevin/hrwros_ws/src/flexbe_app/bin/run_app: line 16:  9855 Aborted                 (core dumped) "$NW" --password-store=basic $ROOT_PATH "$@"
```

After running this command, FlexBE app opens up for a while and then closes.
Now, I am not sure what to do with this.

pschillinger, 2019-05-01:
Sorry for that! I have actually never seen an error like this before... Since this is an image, can you share a link to it?

askkvn, 2019-05-03:
Select the singularity install file (~350 KB) corresponding to the linux distribution you are using (either 14.04 or 16.04 or 18.04)

Singularity for [Ubuntu 14.04](https://courses.edx.org/assets/courseware/v1/474883cf9dca03aeb676694dceb71d15/asset-v1:DelftX+ROS1x+3T2018+type@asset+block/singularity-container_2.5.2_amd64_ubuntu1404.deb)
Singularity for [Ubuntu 16.04](https://courses.edx.org/assets/courseware/v1/a273b8fd47d7f3c2537f631167309720/asset-v1:DelftX+ROS1x+3T2018+type@asset+block/singularity-container_2.5.2_amd64_ubuntu1604.deb)
Singularity for [Ubuntu 18.04](https://courses.edx.org/assets/courseware/v1/8936ebfa0439d58b2a3d5eb3323a2b6e/asset-v1:DelftX+ROS1x+3T2018+type@asset+block/singularity-container_2.5.2_amd64_ubuntu1804_untested.deb)

Singularity Image: https://surfdrive.surf.nl/files/index.php/s/Al2JUbvpQhyvDmZ/download (this file is approximately 1.1 GB)

PS: I am using Ubuntu 18.04 LTS.

pschillinger, 2019-05-21:
I was able to reproduce the initial error by using the image. How did you install the libraries then? It looks like the image is read-only. In general, I have the impression that the image is is intended for the old Chrome version of the app, not this newer version.

askkvn, 2019-05-22:
I have installed libraries by manually adding files to a folder. 
I guess this image is made for teaching purpose only and no write permission given to user. 
By the way, Thanks for your effort and time.

closed 2019-05-22 by askkvn