FlexBE/flexbe_app #37: Problem with linking behaviors and states

AlvaroFerGar, 2019-05-24:
Hello everyone,

Two months ago, I had a perfectly working FlexBE installation. However, I tried to launch it back a few days ago and it doesn't seem to find the behaviours and states packages. 

This is how my **_Configuration_** tab looks like:
[image]

I've removed my custom packages and tried to reinstall it only with the _generic_flexbe_states_ but it neither works. I have followed the tutorial step by step and the packages show no error when building with _catkin_make_  :/

Has anybody experienced anything similar? 

Thanks in advance!!

KuKuXia, 2019-05-29:
Hi AlvaroFerGar. I also experienced this problem now. I can't find the state library panel too.

AlvaroFerGar, 2019-05-29:
> Hi AlvaroFerGar. I also experienced this problem now. I can't find the state library panel too.

Hi KuKuXia!

Unfortunately I haven't figured out what is the problem. I have tested different releases of _flexbe_behavior_engine_  and _flexbe_app_ but nothing has worked :(

pschillinger, 2019-06-01:
Sorry to hear that you face problems!

In order to help you, can you provide some additional information? In
particular, is the ROS workspace sourced in the terminal where you run the
app? Can you roscd to the state packages that are not found? Do the
package.xml files of the state packages contain the export statement?

AlvaroFerGar, 2019-06-02:
Hi  pschillinger
Thanks for your reply!

> ...In particular, is the ROS workspace sourced in the terminal where you run the app? Can you roscd to the state packages that are not found?

I can't check it today but I source my workspace in the _.bashrc_ file and I think that I could _roscd_ to the state packages without any problem. I will post another comment tomorrow if I'm not remembering well.

>...Do the package.xml files of the state packages contain the export statement?

I didn't edit the original downloaded files  so the _export_ tag should be where it's needed.

AlvaroFerGar, 2019-06-03:
Hi again!
I have checked and my workspace is properly sourced, the _.xml_ looks fine and _roscd_ founds all my _flexbe__* packages. I have tried with old releases and with the new ones but nothing appears to work.

Flexbe_app appears to find all the behavior packages int the start.
```
....
[INFO] [1559567857.135413]: 3 behaviors available, ready for start request.
[INFO] [1559567857.668265]: --- Behavior Engine ready! ---
[INFO] [1559567857.669219]: No onboard behavior is active.
[INFO] [1559567857.669896]: --- Behavior Mirror ready! ---
```
I edited the line 34 in io_packageparser.js to see if my state packages are detected with a log message and, surprisingly, yes they are!

After some research I reactivated(in window.html and events.js) the _Force Discover_ button in the Configuration tab. When pressed, "No Behavior Packages" and "No State Packages" messages appear and i can't do anything to solve it. 

Does this give anyone more information? Does someone have a clue of where my flexbe "loses" his related packages?

AlvaroFerGar, 2019-06-03:
I also have found that the _discover_  in  ui/ui_settings.js (line 129) doesn't appear to be executed. The "after" and "before" messages are displayed but not the "inside" one,

```
T.logInfo("After PackageParser");

IO.PackageParser.discover(ros_pkg_cache, (new_packages, added_states, added_behaviors) => {
	T.logInfo("Inside PackageParser");
	ros_pkg_cache = ros_pkg_cache.concat(new_packages);
	state_pkg_cache = state_pkg_cache.concat(added_states);
        
        ....
});
T.logInfo("Before PackageParser");
```

Sorry if this isn't related with my problem. I'm not a js expert :/

AlvaroFerGar, 2019-06-10:
Well, so I've reinstalled ALL my ROS distro and packages and now it seems to work well. I think we can assume that it was a weird bug in my installation :)

Thanks to everyone who tried to help!

pschillinger, 2019-06-12:
Great to hear that it works now and thank you for the above information. I will further investigate to see what was going on. Even if something is weird with the installation, a specific error message should be provided.

pschillinger, 2019-06-14:
The only way I found to achieve the reported observations was the case that one of the ROS packages (not necessarily containing states or behaviors) would have a malformed package.xml. Then, the process of parsing the packages would be stopped due to Javascript's way of handling exceptions in asynchronous background calls.

I added an exception handling for this case now and print a warning message as feedback. This means that even in the case of the error (as long as it's a different package), states and behaviors will be found regardless.

jimmyshe, 2019-09-05:
I followed the tutorial step by step and I got the same error.

luoyuzju, 2019-09-19:
i got the same problem,you can delete the flexbe_app,and redownload it. it seems something wrong about flexbe_app.

jas1-khabrani, 2020-02-20:
[image]

@pschillinger reinstallation does not work !! Please guide

jas1-khabrani, 2020-02-26:
Sloved :
I used the flexbe behavior_engine version 1.2.1 and flexbe_app version 2.2.1 
(works fine it detects all the behaviour )

pschillinger, 2020-03-01:
So do I correctly understand that version 1.2.2 / 2.2.2 do not work, also not after re-installing, but 1.2.1 / 2.2.1 do work as expected?