FlexBE/flexbe_app #38: Behaviour hangs on last state. Forcing doesn't work. Stop Execution doesn't work. And displays the last state name in Lock Behaviour

GalMoore, 2019-07-29:
[image]

GalMoore, 2019-07-29:
[image]

pschillinger, 2019-07-29:
Looks like a parsing error in the ROS interface of the app, I will take a
closer look.

Can you run "rostopic echo /flexbe/command_feedback" during the end of
behavior execution and copy the output here? It would help to see what
triggers the printed warnings. Thanks!

GalMoore, 2019-07-29:
Thanks @pschillinger, I'm not getting anything on rostopic echo /flexbe/command_feedback. I am getting a heartbeat though on rostopic echo /flexbe/heartbeat (even if starts hanging).

GalMoore, 2019-07-29:
> Thanks @pschillinger, I'm not getting anything on rostopic echo /flexbe/command_feedback. I am getting a heartbeat though on rostopic echo /flexbe/heartbeat (even if starts hanging).

anything else I can do to help debug??

pschillinger, 2019-07-29:
I was asking for the topic because of the warning message in your first
post. But then, it might not be related to the issue of hanging.

What is the state where it hangs? Is it one implement by you or one of the
pre-defined states? In the first case, can you share the source code? Is
there anything in the code that could be blocking? In general, any
information that could help to replicate the problem would be useful.

GalMoore, 2019-07-29:
I have a catkin_ws with only  [ROS Openvino installed in it ](https://github.com/intel/ros_openvino_toolkit) I added the three repos from here http://philserver.bplaced.net/fbe/download.php and catkin_make to compile. It downloads dependencies and compiles fine and I source devel... I then use rospack find to find flexbe_app and launch with  roslaunch flexbe_app flexbe_ocs.launch. The app launches and tells me "No Behaviour Packages" - I wasn't sure what to select from the drop down box (is it new? It didn't seem to recognize any behaviors I downloaded ). The drop down box doesn't give any options that mke sense to me so I cloned some[ previous behaviours and states I made a few months ago](https://github.com/GalMoore/robot_brain_behaviours). They seem to be recognized automatically after catkin_make (the previous notice saying "No Behaviours" does not show up) and I can load my behaviours. The states run but fine but always hang on the last state and need to be killed from terminal.

Thoughts?

GalMoore, 2019-07-30:
working on a NUC i7 16GB RAM Ubuntu 18.04

GalMoore, 2019-08-01:
I'm trying to install flexbe from scratch into a new catkin_ws (as I've done before succesfully) but this time I meet this, which I don't remember seeing before
[image]

marekskocz, 2020-06-03:
I am having the same issue here (working on Intel NUC i7 with Ubuntu 18.04 with ROS melodic). I think the problem is somewhere in the link between the user interface and the engine in the background.

The user interface gets stuck once the last state of the behavior is completed, and it is not possible to force stop execution (stopping execution does work as long as the last state is not yet completed). On the top left in the UI it also still says the behavior is 'running'. However, the behavior engine in the background does actually finish the execution of the behavior, as in the terminal I can see messages that the behavior execution is finished and that the behavior engine is ready.

Any ideas how this could be fixed?

JessieMcVicar, 2020-06-15:
I've got the same issue as @marekskocz with Ubuntu 18.04 and ROS melodic

pschillinger, 2020-06-17:
The issue of getting stuck in the last state turned out to be caused by different versions of the yaml package. In fact, depending on how you install it, you get different versions:

* `apt` installs `python-yaml` on latest version `3.12` (I think this is the ROS default for Ubuntu 18.04)
* `pip` installs `pyyaml` on latest version `5.3`

If both are installed, pip wins and causes the above issue.

While I work on fixing the issue itself, you might consider to use the working yaml version:
```
pip uninstall pyyaml
apt install python-yaml  # might be installed already
```

antoan, 2023-03-13:
This doesn't work for me on Noetic, develop branch, unfortunately.

My system python pyyaml is 6.0. I'd be grateful for any workaround.