FlexBE/flexbe_app #42: State list is empty

zengxiaolei, 2019-09-23:
Hello,  the state filter cannot work because it can find no states. Roscd flexbe_states works. 

[image]

 I have also tried reinstalled FlexBE, but problem remains the same.  Available state and behavior is empty in Flex App. I don't known why. Please help.
[image]

Thanks in advance!

pschillinger, 2019-10-18:
Is the workspace sourced in your bashrc? Does this also happen when your run

    roscd flexbe_states
    rosrun flexbe_app run_app -o  # in the same terminal

In general, the way that the FlexBE App looks for state packages is the same that is used for roscd, so I am not sure what might be the problem here. The only difference is that only ROS packages which export "flexbe_states" in their package.xml are listed.

cjue, 2019-10-18:
I once had a similar "states not found" issue after I changed the catkin package name of a flexbe states package. Everything worked fine again after I changed the name back.

To my understanding this happened, because the python module path was different from the catkin package name of my state package and this code relies on them being the same:  
https://github.com/FlexBE/flexbe_app/blob/master/src/ros/ros.js#L108

pschillinger, 2019-10-18:
That's an interesting observation and could indeed be related. I will look
further into this.

zengxiaolei, 2019-10-24:
The workspace is sourced.  This also happens when I follow your steps. 
[image]

zengxiaolei, 2019-11-26:
I uninstalled ubuntu16.04, installed ubuntu18.04 and ROS melodic, now flexbe state is all right.

Sinchiguano, 2019-12-05:
Well, I will leave here my query and also my answer to the problem about this issue.
In melody is working fine. But my goal was to run it in Kinetic. I tried my best to just run the example behavior, but it said that it is unable to find state definition for;
LogState
and WaitState,
If you have any solution about this problem, I would be more than grateful.

//////////////////////////////////////
Answer
////////
Clone the following repo
    flexbe_app
    flexbe_behavior_engine
    generic_flexbe_states
Create a demo using:
rosrun flexbe_widget create_repo ur_robot

Install ros packages by:
sudo apt-get install ros-kinetic-flexbe-*

And finally 
roslaunch flexbe_app flexbe_full.launch

With those steps I was able to use the example behavior in order to test the installation. The goal is to use FlexBe in kinetic since the following 
[image]
state :
https://github.com/CNURobotics/flexible_manipulation
 seems to be suitable for my application. I did try with melody, but i could not compile successfully since it is developed in ROS kinetic.

Best regards,
Cesar

pschillinger, 2019-12-13:
Still not sure what could be the issue for not finding states in kinetic...

Regardless of this, @Sinchiguano I would recommend you to use melodic if possible. @dcconner any idea why *flexible_manipulation* would not work in melodic?

pschillinger, 2020-02-01:
In case it happens again to someone that no state definitions are found, can you please try the following steps:
- Make sure that the FlexBE App is on the latest version, preferably checkout the develop branch
- Go to "Configuration"
- Select the "Python" state parser
- Click "Force Discover"
- Let me know what happens

MariusBeul, 2021-11-29:
I had a similar issue where FlexBE couldn't find the states provided by the flexbe_behavior_engine/flexbe_states package. States provided by other packages (e.g. generic_flexbe_states) were found. The suggestion in https://github.com/FlexBE/flexbe_app/issues/42#issuecomment-581047186 solved the issue.

dcconner, 2023-04-28:
We will be releasing a melodic version of the flexible_manipulation package in the next week or so.  Apologies for missing the earlier tag.

Making note of above comment about need for catkin package and module names to be the same