FlexBE/flexbe_app #43: ERROR:input_method_base.cc

yubaoliu, 2019-11-03:
`/catkin_ws$ roslaunch flexbe_app flexbe_full.launch
... logging to /home/yubao/.ros/log/fb537152-fdd1-11e9-a8ce-309c23e1152f/roslaunch-yubao-Z370M-S01-6367.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.15.23.26:40533/

SUMMARY
========

PARAMETERS
 * /behavior/log_enabled: False
 * /behavior/log_folder: ~/.flexbe_logs
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    behavior (flexbe_onboard/start_behavior)
    behavior_launcher (flexbe_widget/be_launcher)
    behavior_mirror (flexbe_mirror/behavior_mirror_sm)
    flexbe_app (flexbe_app/run_app)

auto-starting new master
process[master]: started with pid [6378]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fb537152-fdd1-11e9-a8ce-309c23e1152f
process[rosout-1]: started with pid [6395]
started core service [/rosout]
process[behavior_mirror-2]: started with pid [6405]
process[flexbe_app-3]: started with pid [6416]
process[behavior_launcher-4]: started with pid [6418]
process[behavior-5]: started with pid [6419]
[INFO] [1572741395.235183]: 1 behaviors available, ready for start request.
[INFO] [1572741395.745634]: --- Behavior Engine ready! ---
[INFO] [1572741395.746462]: No onboard behavior is active.
[INFO] [1572741395.747591]: --- Behavior Mirror ready! ---
[6420:6420:1103/093643.174381:ERROR:input_method_base.cc(146)] Not implemented reached in virtual ui::InputMethodKeyboardController *ui::InputMethodBase::GetInputMethodKeyboardController()Using InputMethodKeyboardControllerStub
`
When launch
`roslaunch flexbe_app flexbe_full.launch`

No package list here:
[image]

No State listed:
[image]

I cloned these to my catkin_ws
1. flexbe_app
2. flexbe_behavior_engine
3.  generic_flexbe_states

And create a demo using:
`rosrun flexbe_widget create_repo tutorial`

And install ros packages by:
`$ sudo apt-get install ros-kinetic-flexbe-*`

`roscd flexbe_states` is no problem.

Here is my system info:
`$ uname  -a
Linux yubao-Z370M-S01 4.15.0-66-generic #75~16.04.1-Ubuntu SMP Tue Oct 1 14:01:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
`

Thanks

pschillinger, 2019-11-19:
I am not sure about the "ERROR:input_method_base.cc", I think this comes from nwjs internally and I have seen this messages like this independently from no states being listed.

In fact, I think your problem could be more related to #42, but I have no idea yet what could be the reason since I did not yet manage to reproduce the error.

dcconner, 2023-04-28:
closing as stale.

closed 2023-04-28 by dcconner