FlexBE/flexbe_app #47: Runutime Control Display bug

zengxiaolei, 2019-12-13:
What I have done is as follows:
 1. Use flex_app to run behavior_1
 2. After bahavior_1 is completed, keep flexbe_app opening, but  run behavior_2 by calling action (http://wiki.ros.org/flexbe/Tutorials/Running%20Behaviors%20Without%20Operator).
3. After behavior_2 is completed, run behavior_1 again by flexbe_app. Behavior works, but no runtime control display in the app. As shown by Fig. 1, on-board status is running and  initialization is done, which is really done. Since the display is empty, I cannot continue the next state since a high-level outcome is to be verified as shown in Fg. 2.

[image]
Fig. 1

[image]
Fig. 2

I tried to stop Execution and restart behavior for many times, but problem remains.

pschillinger, 2019-12-13:
Thanks for reporting, I can reproduce this bug and will look into it.

pschillinger, 2019-12-13:
It turned out that the FlexBE App worked as intended, but the issue was caused by flexbe_mirror due to improper release of its thread lock. Consequently, it did not send any behavior updates to the App afterwards.

I fixed it with the above commit.

zengxiaolei, 2019-12-18:
It works!

pschillinger, 2020-02-01:
Fixed in flexbe_behavior_engine version 1.2.3

closed 2020-02-01 by pschillinger