FlexBE/flexbe_app #48: states lost in version 2.2.2

zengxiaolei, 2020-01-12:
Firstly, It's nice to see a new version of flexbe_app and flexbe_behavior_engine :-)
I checkout flexbe_behavior_engine ( version 1.2.3) and flexbe_app (2.2.2), some states of my own package are lost, as shown by Fig.1. Only one state is left. 
Then I checkout  flexbe_app (2.2.1) back,  flexbe_behavior_engine is still in new version 1.2.3, states are complete, as shown by Fig.2.
[image]
Fig. 1

[image]
Fig. 2

Besides, Behaviors are always available, but they cannot work since some states disappear.
Will the new version be incompatible with the custom states?

pschillinger, 2020-01-12:
Sorry for that, this was clearly not intended! Indeed, the way how packages
are parsed changed slightly so that the progress can be tracked and also
visualized. Furthermore, a new alternative state parser has been added, but
is not selected by default. In any case, all states should still be
compatible and nothing changed in the expected format.

When I tested it locally and on CI based on the flexbe_states package and
some custom test packages, I didn't observe issues in recognizing the
states. Is there anything you are aware of that could be different in your
package? Would it be possible to share the state package so that I can take
a look myself?

zengxiaolei, 2020-01-12:
I cannot be aware of the difference... Even the only left state "BehaviorGate" has nothing special. Here is the package:  https://github.com/zengxiaolei/agv_behaviors      
You can try yourself, please.

pschillinger, 2020-01-12:
Should be fixed by the above commit. It is not yet merged into master, but I will do a new release as soon as I have a more stable internet connection than on this train.

Problem was that you had states that have an entry in their documentation that is not correctly formatted, see e.g. diehl_a_or_b.py where `<= done` is stated in the docstring, but no text follows. This is normally not an issue, but will print a warning so that it can be tracked why there is no documentation for this outcome. Now, the bug here was that there was a failure during printing of this warning that stopped the further state parsing. This is fixed now and in addition robustified against similar problems.

zengxiaolei, 2020-01-12:
I tested your new commit, It works. Thanks a lot for your immediate reply.
Besides, I should and will make my  state doc-string normative ;-)

pschillinger, 2020-02-01:
Has been fixed in 2.2.3

closed 2020-02-01 by pschillinger