FlexBE/flexbe_app #49: binary not released to apt

wmmc88, 2020-01-16:
It looks like `flexbe_app` is not available to install through apt on ubuntu. This is true for both kinetic and melodic. As you can see, they are [not listed](http://repositories.ros.org/status_page/kinetic_default.html?q=flexbe). I believe the docs are failing to build because of dependencies and this is preventing the release of the binary. The builds fail [here](http://build.ros.org/job/Mdoc__flexbe_app__ubuntu_bionic_amd64/2/console#console-section-9) and [here](http://build.ros.org/job/Kdoc__flexbe_app__ubuntu_xenial_amd64/4/console)

pschillinger, 2020-01-19:
Yes, the package is indeed not released as a binary via apt. But this is because of the issue mentioned in ros/rosdistro#19823. In short, this causes the debian linter to complain about the way how nwjs (the underlying framework of the app) and catkin work together. I did not look further into this, but any help would be appreciated.

Why the documentation build is failing, I don't know yet. In the more recent build it says:
```
4:23:58 flexbe_app has BUILD message dependencies:
04:23:58 flexbe_app has BUILD dependencies:
04:23:58   - rostest
04:23:58 -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
04:23:58 CMake Error at /opt/ros/melodic/share/flexbe_onboard/cmake/flexbe_onboardConfig.cmake:173 (message):
04:23:58   Project 'flexbe_app' tried to find library 'flexbe_onboard'.  The library
04:23:58   is neither a target nor built/installed properly.  Did you compile project
04:23:58   'flexbe_onboard'? Did you find_package() it before the subdirectory
04:23:58   containing its code is included?
04:23:58 Call Stack (most recent call first):
04:23:58   CMakeLists.txt:21 (find_package)
04:23:58   CMakeLists.txt:97 (select_catkin_dependencies)
04:23:58 
04:23:58 
04:23:58 -- Configuring incomplete, errors occurred!
```
However, a few lines above it also says:
```
04:23:18 Setting up ros-melodic-flexbe-onboard (1.2.3-1bionic.20200116.142320) ...
```

Indeed, [in this commit](https://github.com/FlexBE/flexbe_app/commit/f699af770262dbf8a6eb8c7ebb91206b84cf480c#diff-34ae68f4adad56c25c5bc05dcb64794e), the dependency has recently been added. However, it should only be a run dependency and is not used during the build procedure. Not sure why it tries to access it and also not sure why it doesn't find it after correctly installing it. In any case, it feels wrong to add it as build dependency.

dcconner, 2023-07-24:
Confirming that flexbe_app will not be release as binary in the near term .  See #90

closed 2023-07-24 by dcconner