FlexBE/flexbe_app #53: Flexbe_app isn't usable anymore

my-name-is-D, 2020-02-07:
hello, this new flexbe_app version is not usable anymore. 
The flexbe_app open correctly but the manifest of the behavior packages aren't seen anymore. (we can't create anything).

I fixed the problem in my computer (after 2 days of strugling) by downloading this version and changing the run_app (in the bin) by a previous version of your flexbe_app (the 2.0.6 i could find on ros_ignite) and adding some files.

The 2.0.6 version isn't sustainable because the nwjs version is not recent enough and the behaviors can't be saved. So i did a mix with both versions (recuperated the manifest file,  lib and locates files and changed the "app_run")
I still get the error "Your profile can not be used because it is from a newer version of NW.js.
Some features may be unavailable. Please specify a different profile directory or use a newer version of NW.js."

But now i can actually see my behaviors files and save them.

The weird mix done isn't a good solution, it would be great to fix this please :) 
Here is the mix i did.

Thank you for your great app!
[run_app.txt](https://github.com/FlexBE/flexbe_app/files/4171848/run_app.txt)

my-name-is-D, 2020-02-07:
the issue : https://github.com/FlexBE/flexbe_app/issues/37

is about the same problem. It seems to be some change done to the version before may 2019.

my-name-is-D, 2020-02-07:
Well, probably due to my little mix-up :
 [ERROR] [1581098093.424571]: Exception caught in behavior definition:
No module named tutorial_flexbe_states.my_py_name

it can't find the py made. 
(it works perfectly in the version 2.0.6 -ros ignite- but not in the new version.)

pschillinger, 2020-02-09:
Thanks for the detailed description of the problem and the steps that you took.

Can you let me know what exactly you changed to make it (somehow) work, e.g.

> I fixed the problem in my computer (after 2 days of strugling) by downloading this version and changing the run_app (in the bin) by a previous version of your flexbe_app (the 2.0.6 i could find on ros_ignite) and adding some files.

Which version do you use and which other files did you change or add?

Also, it would be helpful to get some more information about your workspace. If the `run_app` change helped to make it work, what would `rospack find flexbe_app` return and which files are located at the given path?

Do you use custom states and behaviors or does it also happen when you only have standard packages and one created by `rosrun flexbe_widget create_repo`?

FrGe2016, 2020-02-21:
Hi I just installed flexbe using the instruction of this page ( Binary at the first step ) and the only available option for the other steps. My installation is on ROS Melodic Ubuntu 18.04.

I also copier two packages that were working perfectly in the Ignite academy course that i just finished. The hole thing contains:
 - the automaticaly generated example_behavior_sm.py + a few simple homade behaviors using proxypublisher or ProxyActionClient and a few wait_time states

After launch i have the following error messages:
[25764:25764:0220/224220.544182:ERROR:component_loader.cc(164)] Failed to parse extension manifest.
(node:25810) [DEP0025] DeprecationWarning: sys is deprecated. Use util instead.

Then.
[25764:25764:0220/224220.544182:ERROR:component_loader.cc(164)] Failed to parse extension manifest.
(node:25810) [DEP0025] DeprecationWarning: sys is deprecated. Use util instead.

When the gui opens,
      - i can not select any packages that contains my behaviors
      - the only package in the drop_down menu is a package contains neither behavior or states
           ( it is the first package in my workspace in alphabetical order ?? )

If it can help

my-name-is-D, 2020-02-25:
Sorry for the late answer. 

I can't tell you those additional information because i redid my whole ubuntu. 
Flexbe has been installed (as prescribed on the Flexbe website) on 3 kinetic and 5melodics, only one was successful. 

For all the others the problem is always the same:  no behavior presented, as FrGe2016, and when it opens a yellow window pop up on the right side of the application. (sorry no preview at the moment) asking to add a behavior (if i recall correctly). i remember a warning sign on it. and on the behavior side, no package proposed.

The weird manipulation i did previously was to install the 2.0.6 version i found (was working on a simulator) and corrected all the errors and problems with the files in the newer version. 
It wasn't actually a good solution because of compatibility problem. Sorry.

For FrGle2016, just checked on the computer that got Flexbe running, it also has:
[2789:2789:0225/144431.851250:ERROR:component_loader.cc(164)] Failed to parse extension manifest.
(node:2894) [DEP0025] DeprecationWarning: sys is deprecated. Use util instead.

but it works fine. 
To note: it didn't work fine at first either, the computer was restarted multiple times before it did! I have no idea why...

The problem may be elsewhere but i can't seem to find it... I don't find any difference between the pc running it and the 7 other servers! (actually it's 4pc with multiple boots). 

It would be great if you knew of those instances to resolve it!

ZakariaChekakta, 2020-03-04:
Hi, I want to use that app but the latest version of it still not working on ubuntu 18.04 and ros melodic, is there is any solution in the near future?

pschillinger, 2020-03-04:
I hope there will be a solution. I'm working on it, but still not able to reproduce the problem.

> After launch i have the following error messages:
> [25764:25764:0220/224220.544182:ERROR:component_loader.cc(164)] Failed to parse extension manifest.
(node:25810) [DEP0025] DeprecationWarning: sys is deprecated. Use util instead.
>
>Then.
>[25764:25764:0220/224220.544182:ERROR:component_loader.cc(164)] Failed to parse extension manifest.
(node:25810) [DEP0025] DeprecationWarning: sys is deprecated. Use util instead.

Those errors result from nwjs. While I have no control over them, I get similar ones and it still works. As long as the window opens, they should not be related.

> For all the others the problem is always the same: no behavior presented, as FrGe2016, and when it opens a yellow window pop up on the right side of the application. (sorry no preview at the moment) [...]

This yellow window always appears if no behavior package is detected and it lists all detected packages in the workspace (not installed). Let me know in case this list is incomplete, i.e.,

>  the only package in the drop_down menu is a package contains neither behavior or states
( it is the first package in my workspace in alphabetical order ?? )

Is this package the only one in your workspace? Or are there any packages missing? If some are missing, what's the first missing one in alphabetical order?

For trying again, can you please use the `flexbe_app` repo on the latest `develop` branch and do the following:

* In a terminal, run `rospack list | wc -l` and remember the number
* Run the app and press `Ctrl+T` to open the internal terminal
* Look for an output like "No new ROS packages detected (xxx in cache)." or "Checking xxx ROS packages for states and behaviors  (xxx in cache)..."
* Does the number in cache (plus new packages if any) match the remembered number?
* In any case, keep the terminal open, go to the "Configuration" view and click on "Force Discover"
* A new output like above should appear (0 packages in cache), are the missing packages now detected?

FrGe2016, 2020-03-04:
Hi

My problem is partly solved with the force Discover command

But lets give feedback step by step
Rospack list gave 550 packages
The fisrt look at the console gave nothing looking like the suggested output
Only a list of packages in red containing 
       Some behavior packages (that i had moves to another directory many days ago when i made     
       the first installation )
       one package that has nothing to do with behaviors or flexbe ( actions_quiz)

After i have done the  "Force discover" 
       The 2 package from my course have apeared in the workpace section of the config pannel and 
        as  choice in the behavior dashboard.
        But:
              - the_virgir_beaviors that i have cloned have not 
                         When i look of the directory structure of this package it is very different compared to 
                          the 2 packages that are displayed

              **- the actions_quiz is still in apearing in the workspace 
                but also has the Default_package in code generation**
         
 Other good new, if i stop and start the app, my 2 packages are available for selection but the above problems remains.

cjue, 2020-03-23:
In my recent experience the Python parser fails to parse some states when they are missing action/service/message dependencies.

An example for this would be the ExampleActionState that tries to import an action from `chrores.msg`: http://wiki.ros.org/flexbe/Tutorials/Developing%20Actionlib%20States

In the current develop branch this leads to a useful warning about the import error in the FlexBE App Gui Terminal. Sill, the behavior differs from the old regex parser, which would happily parse and allow the addition of states from files with broken imports.

So, @FrGe2016, do you get any error output in the app after pressing Force Discover, with the current `develop` branch of the app?

pschillinger, 2020-03-31:
> Sill, the behavior differs from the old regex parser, which would happily parse and allow the addition of states from files with broken imports.

Yes, this is a difference by design. The two parsers are not exactly the same. While the Python parser is supposed to support a wider range in what can be done (e.g., sub-classing of states), the Regex parser is more independent from how Python works (e.g., can ignore import errors).

> the_virgir_beaviors that i have cloned have not

vigir_behaviors still has the package structure for the old Chrome App, see [Backwards Compatibility](https://github.com/FlexBE/flexbe_app#backwards-compatibility).

**Note:** The recently released version 2.2.4 contains several improvements how the ROS package cache is used and also adds an option to disable it. After updating, please click "Force Discover" once to reset the cache. If you still face any issues, let me know, disable the cache, and check whether the issue persists.

evangelosc, 2020-05-14:
OS: Ubuntu 18.04
ROS: Melodic
Flexbe_app: 2.2.1
Flexbe_behavior_enginer: 1.2.1
nwjs: 0.34.1

**LOG**
[10395:10426:0514/160645.243927:ERROR:web_database_backend.cc(113)] Cannot initialize the web database: 2
[10395:10395:0514/160645.292314:ERROR:desktop_window_tree_host_x11.cc(1125)] Not implemented reached in virtual void views::DesktopWindowTreeHostX11::InitModalType(ui::ModalType)
[10395:10410:0514/160645.468224:ERROR:top_sites_backend.cc(92)] Failed to initialize database.
INFO: cannot create a symlink to latest log directory: [Errno 17] File exists
[10395:10409:0514/160645.900218:ERROR:simple_version_upgrade.cc(164)] Inconsistent cache version.
[10395:10409:0514/160645.900231:ERROR:simple_backend_impl.cc(710)] Simple Cache Backend: wrong file structure on disk: /home/evangelos/.cache/FlexBE App/Default/Cache

I got the above error message on terminal when I started the execution of my behavior and flexbe_app hangs. The execution of the behavior does not finish and the stop execution option is not clickable. Any help would be appreciated.

evangelosc, 2020-06-03:
@pschillinger is there any update on it?

evangelosc, 2020-09-08:
OS: Ubuntu 16.04
ROS: Kinetic
Flexbe_app: 2.2.1
Flexbe_behavior_enginer: 1.2.1
nwjs: 0.34.1

**LOG**
[ERROR] [1599578874.550937]: Exception caught in behavior definition:
No module named msg
[ERROR] [1599578874.551463]: Dropped behavior start request because preparation failed.

When I run a behavior twice, it throws this exception on the terminal. What does it trigger it? Any idea?
Is there any way to discard it from the source code?

cjue, 2020-09-11:
@evangelosc: 

When you say "run a behavior twice", do you mean pressing "Start behavior" again after the first one completed?

Did you try the newer versions 2.2.4/1.2.4, or the current develop branch? These work very well for me on 16.04.

evangelosc, 2020-09-11:
@cjue Thanks for your response. What's your nwjs version?

cjue, 2020-09-11:
My nwjs/version is 0.42.3

dcconner, 2023-04-28:
#71 discussion of parser issues

As this discussion is a bit stale, I'm inclined to close this issue unless folks have additional input

PanithanRit, 2024-01-17:
Hi I try to use FlexBe on 17 Jan 2024. Three day agon I still can use FlexBe. Apparently, To day I try to install FlexBe again and it doesn't work anymore. I use ubuntu 20.04 LTS with ROS noetic. This is the problem 
[image]

I try to delete ROS uninstall ubuntu it still get the same problem. I use FlexBe noetic branch and It did work before three day ago.  I try not to update ubuntu 20.04 but in order to install ROS I need to use sudo apt update.  The flexBe onboard packages are working fine but flexbe_app doesn't work. If this problem can't be fix is there other way to create behavior with out UI ?