FlexBE/flexbe_app #54: checkBehavior as command line tool

benmaidel, 2020-02-11:
@pschillinger Do you see any easy way to get the checkBehavior functionality from flexbe_app accessible from the command line? So we could do something like `rosrun flexbe_app check_behavior behavior_name`

Background: We'd like to add the checkBehavior feature from flexbe_app to the rostest pipeline.

pschillinger, 2020-02-11:
I agree that this would be a helpful feature. Unfortunately, I don't think that there is an "easy" way to achieve this. Main reason is that the particular "Check Behavior" function offered by the App includes a number of checks that rely on a complete model of the behavior, which requires most of the App's functionality (e.g., for resolving variables, verifying state interfaces, checking all possible execution paths for missing userdata).

Still, there would be some alternatives that appear to be more in reach and could already be helpful:
* `flexbe_testing` now also supports behavior tests. While this is meant to execute behaviors, the `import_only` version is already enough to verify that all required states can be imported. I think this should be a must-have for behaviors in any CI pipeline.
* If you are ok with actually running the App in your CI (see `flexbe_ci` [for an example](https://github.com/FlexBE/flexbe_ci/blob/51096dfc4f1f0cc590fa862bc4ce36381073e2bc/ci_scripts/script.bash#L19)), it should be possible to automate the "Check Behavior" button. I would still need to implement this, but it should be fairly easy.
* A "Check Behavior" script that is independent of the App could be implemented in Python. This is a bit more work and I cannot yet promise that every single check can be covered, but might still be helpful.

Does any of these options sound helpful to you?

benmaidel, 2020-02-26:
@pschillinger thank's for your feedback. I've already suspected that it will go this way. We are already using the `import_only` tests from `flexbe_testing` and they help a lot. For your second solution I've prepared a PR https://github.com/FlexBE/flexbe_app/pull/55 with a draft solution. Would be great if you could take a look at it.

pschillinger, 2020-03-01:
Thanks for already preparing a PR! I will take a look.

pschillinger, 2020-03-01:
PR looked good, I just added a bit more error handling to prevent the App from crashing on invalid behaviors and merged it. In case you have any additions, feel free to re-open a PR.

I also changed the way how the report is parsed so that all behaviors are reported as separate tests and in case of failure, the error message generated by the App is shown by rostest.

pschillinger, 2020-03-31:
Release with 2.2.4

closed 2020-03-31 by pschillinger