FlexBE/flexbe_app #56: Extreme CPU load by Python processes after shutdown of FlexBE app

joostvanstuijvenberg, 2020-03-06:
While using the FlexBE app through `roslaunch flexbe_app flexbe_ocs.launch`, I can see in output of **top** that a bunch of Python-processes are running (probably 10 or something). When I close the app these processes start to consume way more CPU cycles than before (on average 0,3 while using the app, around 75 right after shutdown.
[image]
[image]

pschillinger, 2020-03-06:
Thanks for updating! I was about to ask which branch you used because the issue should be fixed by [this commit](https://github.com/FlexBE/flexbe_app/commit/a927e1d8751825a4c3f031b2d9d6f63a4fda435b) on develop. I plan to merge it into master during the next days.

joostvanstuijvenberg, 2020-03-06:
Great! Keep up the good work.

In the mean time I'll have some reading material that might somehow feel familiar to you :-)

Cheers,
Joost

pschillinger, 2020-03-31:
Released with 2.2.4

closed 2020-03-31 by pschillinger