FlexBE/flexbe_app #59: repeated behavior and package display

zengxiaolei, 2020-03-19:
Hi, 
first, what I use is ROS melodic in Ubuntu18.04,  flexbe_app develop branch (latest) and flexbe_behavior_engine  develop branch (latest).  When I load behaviors in flexbe APP, all available behaviors and the available package display twice as shown in Fig. 1.
[image]
Fig.1

The states display are normal.The following are references that may be useful.
[image]
Fig.2

[image]
Fig. 3

More information can be provided if needed.

pschillinger, 2020-03-22:
This might be an issue with the ROS package cache that the FlexBE App uses internally to not check all packages for states and behaviors during every start. Can you go to the "Configuration" page and press "Force Discover"? Does this fix it? It might take a moment to refresh the cache.

Regarding you question, did you check [this tutorial](http://wiki.ros.org/flexbe/Tutorials/Running%20Behaviors%20Without%20Operator)? If this does not provide the function you need, what exactly would you need? In addition to what is written there, the Python API exists for looking up behaviors by name, but there is no command-line script for it (yet).

zengxiaolei, 2020-03-26:
"Force Discover" works! Thanks a lot.

closed 2020-03-31 by pschillinger