FlexBE/flexbe_app #61: reload/refresh state modules from Statemachine Editor function

GerardHarkemaAvans, 2020-04-11:
As mentioned in #10 there should be a function to update the states after changing ist python script. Sorry, but i cannot find this function in the user-interface of the flexbe app. What is the correct procedure to update the design?

Thanks in advance,

Gerard

pschillinger, 2020-04-17:
The referenced update function is automatic and works in the background. Using Ubuntu's mechanism to receive updates about changed files, the state files are automatically re-loaded when such notifications are received. No need to do something manually, just save the state's python file.

asierfernandez, 2020-04-24:
We are experiencing the same issue. We've tested it under Ubuntu 16.04 and Ubuntu 18.04, in both cases with **flexbe_app version 2.2.4**

If we edit a python script and save it, flexbe_app automatically detects it and its integrated `console` prompts and through this error:
```bash
ERROR: ENOENT: no such file or directory, open
`path_to_state.py.xxx____xxxx.py`
```
Even that error, the application can be launched, but changes made in the script are not available.  We need to re-launch flexbe_app.

If more info is required, please let me know.

pschillinger, 2020-04-26:
I indeed found an issue that prevented the automated state reloading when using the Python parser: Since Python internally caches any previously imported module and does not reload it again, the state did not update despite being triggered correctly. This was not an issue for the (default) regex parser and should be fixed for the Python parser by the above commit. Does this help?

The file path `path_to_state.py.xxx____xxxx.py` from the error message looks weird, not sure where this could come from. What would be the correct path?

asierfernandez, 2020-04-27:
Even applying aa61ea9efcf237472dcb0a0c59fa5411b2de8c5f, I continue having the same behavior.

I've figured out what `path_to_state.py.xxx____xxxx.py`is. 
`path_to_python_script.py` + `md5sum of previous path` + `.py`

I hope it helps!

joao-aguizo, 2021-11-22:
I think this issue was caused by this hotfix: https://github.com/team-vigir/flexbe_behavior_engine/commit/bbe82cf28392ad82d09cce07cc6210579c91b6e0, since only after applying it to an older version of flexbe_behavior_engine that I am using (https://github.com/team-vigir/flexbe_behavior_engine/tree/1.2.5), I have started watching the aforementioned issue in the aforementioned posts.

UNRELATED: I had to apply this hotfix so that I can consistently run some behaviors. That being said, had I not applied this fix I would only be able to run some behaviors that rely on services once, being that the second time would fail with the following error:
`ImportError: No module named srv`

dcconner, 2023-04-28:
Any status updates on this?  Are people still experiencing issues? 

If I understand correctly, you are editing the python state implementation files, and want those changes reflected when you run a behavior without rebuilding and restarting FlexBE. 

I presume this is on a single machine running from devel in ROS 1.