FlexBE/flexbe_app #63: Can't edit sub-behavior state's parameters

asierfernandez, 2020-04-23:
We want to encapsulate some states within a behavior in order to reuse them in different applications/projects.

States within this behavior have some parameters we want to change depending the application. The thing is we can only change parameters when we design the behavior, not in the application.

I don't know if I'm explaining well enough. Let's take a simple example:

1.  We create a behavior containing a wait state and a logger state.   In this behavior we can modify for either **wait_time** or logger **text** parameters.

[image]

2. We create a new behavior containing 2 previously created behaviors.

[image]

I'd expect if I enter in each behavior (wait_behavior_1, and wait_behavior_2) I could change all **wait_time** and **text** parameters, but I can not. 

I might missing something, or doing something wrong. For me, this should be a core functionality for scalability of application and projects.

In this example, an alternative is to encapsulate the wait and logger states within a state-machine container, and copy/paste to duplicate them, but I don't think this is the way to go.

Any idea?

Thanks a lot for making such a tool available. Even if we just started using it, it looks very promising!

pschillinger, 2020-04-26:
Your use case should be covered by the _Behavior Parameters_ (thats the box to the bottom left in the behavior dashboard). In the above example, you could do the following:

1. Add a Behavior Parameter to your behavior "wait behavior", let's call it "wait_time". You can configure it by using the pencil icon next to it, but this is not required here.

2. In the editor of "wait behavior", change the parameter value of the "wait" state to use "self.wait_time" instead of a hard-coded value. If you start typing "wait_time", auto-complete should propose the parameter. Apply the changes and save the behavior.

3. In your behavior "multiple wait behaviors", single-click the respective behavior so that the properties panel opens. You should see an entry "wait_time" listed as parameter. Uncheck the default and enter your desired value.

Entering a behavior is only meant as a read-only view of the behavior that shows what is inside. The content is to be changed only when editing the included behavior itself because otherwise, changing it at one included place would also change it in all other included places.

asierfernandez, 2020-04-27:
Thanks for the information, very clear!
@pschillinger you can close the issue from my side.

closed 2020-06-05 by pschillinger