FlexBE/flexbe_app #64: Can man  save behavior as another with different name?

zengxiaolei, 2020-06-05:
Hi,  first thanks for your work and I really like to use this wonderful tool .

Sometimes I would like to create two similar behavior but with some difference, but I  need to create two new behaviors with some repeat work.  If the behavior_1 can be saved as behavior_2, then I can load behavior_2 and only edit where is different.  This can be easy for users. Is it reasonable?

pschillinger, 2020-06-05:
Yes, this is totally possible. You can simply create behavior_1 (or just the common part of the two), save the behavior, change the name to behavior_2, and save again. You will now have the two behaviors and can load and edit them independently. Still, they are not "formally" coupled together, so after you saved it under the second name, you will need to do any future work twice (or simply copy&paste the part to the other).

zengxiaolei, 2020-06-08:
Yes, This is also very convenient ! Thanks!

zengxiaolei, 2020-06-08:
By the way, how about delete some behaviors? What I do is to delete the associated .xml file and _sm.py file. I think this is less convenient.

pschillinger, 2020-07-24:
Deleting those two files is correct. I agree that it is less convenient, but did not add a "Delete" button yet. Will consider this for the future.

closed 2023-04-28 by dcconner