FlexBE/flexbe_app #65: State parameter pre-defined options imported incorrectly

tcassell94, 2020-07-02:
If you set some pre-defined options for parameter values (such as in [this example state](https://github.com/team-vigir/vigir_behaviors/blob/master/vigir_flexbe_states/src/vigir_flexbe_states/change_control_mode_action_state.py)), they are accessed incorrectly in the behavior containing the state.

IE, the state itself would be imported as `from flexbe_states_package.state_name import StateName` in the behavior_name_sm.py file. But then the state's default parameters are accessed as `flexbe_states_package.StateName.DefaultParameterName`.

I think the fix is to just access the default parameters as `StateName.DefaultParameterName`.

tcassell94, 2020-07-03:
It's caused by the value of the default parameter being set in the FlexBE App as `package_name.StateName.DefaultParameterName` when you click your desired parameter value from the drop-down options. So I guess when you click the drop down the parameter should be set to `StateName.DefaultParameterName`.

pschillinger, 2020-07-24:
This actually depends on whether the state class name is a duplicate in the workspace or if the respective setting is enabled to consider the state package in addition to the class in general. Still, the completion you pointed out is incorrect in both cases.

The above commit fixes it for both of them.

commit 7eb0f981 (2020-07-24): Use correct state prefix in autocompletion
(fix #65)
closed 2020-11-19 by pschillinger