FlexBE/flexbe_app #66: "Failed to call service: transport error completing service call" when behavior execution is started second time

yfedi, 2020-07-13:
Hello,

**Note: This issue doesn't occur when you start behavior execution first time after launching flexbe_app GUI**

We have implemented a state machine using flexbe and during testing, we found the next issue occurs when ProxyServiceCaller is used:
`Failed to call service: transport error completing service call`

Typical workflow to reproduce the issue:
1. Launch needed  service and flexbe_app(roslaunch xv_flexbe_behaviors flexbe_full.launch)
2. Load needed behavior and start execution. Go to the state where service is called. A call to service is executed without any issues.
3. Stop execution behavior and start it again. Go to the state where service is called. 
**Expected behavior:** A call to service is executed without any issues.
**Actual behavior:** A call to service is not executed properly. Service received a request and send a response, but the response is not deserialized by **ProxyServiceCaller**. 
This issue is not related to specific service, the same error occurs when we call any other service.

To find the root cause of the issue, we eliminated all other code and create a state machine with a single state, where we call service using ProxyServiceCaller and using ServiceProxy from rospy. All source code and log you can find below

As you can see from logs **ServiceProxy from rospy**  works in the same time without any issues.

Log with issue reproducing:
```
[3:36:52 PM] --> Starting new behavior...
[3:36:52 PM] ProxyServiceCaller
[3:36:52 PM] ProxyServiceCaller call status: success: True [+]
[3:36:52 PM] Trying rospy call
[3:36:52 PM] rospy call status: Hey, roger that; we'll be right there!
[3:36:52 PM] Finished
[3:36:52 PM] No behavior active.
[3:36:54 PM] --> Starting new behavior...
[3:36:55 PM] ProxyServiceCaller
[3:36:55 PM] [Demo] Failed to call service: transport error completing service call: receive_once[/fake_911]: DeserializationError cannot deserialize: 'NoneType' object has no attribute 'error'
[3:36:55 PM] Trying rospy call
[3:36:55 PM] rospy call status: Hey, roger that; we'll be right there!
[3:36:55 PM] Finished
[3:36:55 PM] No behavior active.
```

State machine code:
```python
from flexbe_core import Behavior, Autonomy, OperatableStateMachine, ConcurrencyContainer, PriorityContainer, Logger
from xv_flexbe_states.service_call_test import ServiceCallTest

class testSM(Behavior):

	def __init__(self):
		super(testSM, self).__init__()
		self.name = 'test'

	def create(self):
		# x:32 y:643, x:130 y:643
		_state_machine = OperatableStateMachine(outcomes=['finished', 'failed'])

		with _state_machine:
			# x:30 y:132
			OperatableStateMachine.add('Demo',
									ServiceCallTest(),
									transitions={'done': 'finished', 'error': 'failed'},
									autonomy={'done': Autonomy.Off, 'error': Autonomy.Off})
		return _state_machine
```

State code:
```python
#!/usr/bin/env python

from flexbe_core import EventState, Logger

from flexbe_core.proxy import ProxyServiceCaller
from std_srvs.srv import Trigger, TriggerRequest
import rospy

class ServiceCallTest(EventState):

    def __init__(self):
        super(ServiceCallTest, self).__init__(outcomes=['done', 'error'])

        self._topic = "/fake_911"

        self._client = ProxyServiceCaller({self._topic: Trigger})
        self._caller = rospy.ServiceProxy(self._topic, Trigger)
        self._error = False
        self._result = True

    def execute(self, userdata):
        if self._error:
            return 'error'

        if self._result is not None:
            return 'done'

    def on_enter(self, userdata):
        Logger.loginfo("ProxyServiceCaller")
        try:
            self._result = self._client.call(self._topic, TriggerRequest())
            Logger.loginfo("ProxyServiceCaller call status: {}".format(self._result))
        except Exception as e:
            Logger.logwarn('[{0}] Failed to call service: {1}'.format(self.name, str(e)))

        Logger.loginfo("Trying rospy call")
        try:
            resp1 = self._caller(TriggerRequest())
            Logger.loginfo("rospy call status: {}".format(resp1.message))
        except rospy.ServiceException as e:
            Logger.logerr("Service call failed: {}".format(str(e)))

        Logger.loginfo("Finished")
```

Service code
```python
#! /usr/bin/env python

import rospy
from std_srvs.srv import Trigger, TriggerResponse

class TestService():

    def __init__(self):
        
        rospy.init_node('xv_order_manager')
        my_service = rospy.Service('/fake_911', Trigger, self.trigger_response)

    def trigger_response(self, request):
        try:
            rospy.logerr("Service call {}")
            return TriggerResponse(
                success=True,
                message="Hey, roger that; we'll be right there!"
            )
        except rospy.ServiceException, e:
            rospy.logerr("Service call failed: {}".format(e))
  

if __name__ == "__main__":   
    ts = TestService()
    rospy.spin()
```

fgrcar, 2020-07-22:
I have the same issue. On the first execution of a behavior it works, on the later ones I receive that same error.

pschillinger, 2020-07-24:
Thank you for the detailed bug report and the code to reproduce. I was already chasing a similar issue, both likely caused by a feature added with version 1.2.5 (see [this commit](https://github.com/team-vigir/flexbe_behavior_engine/commit/dd5fc2d6c01459bd1e06e05eb861fd209e839a96)), but couldn't reproduce it so far. With your code I could.

I removed this new feature again now in the above commit and this should fix the issue for you.

closed 2023-04-28 by dcconner