FlexBE/flexbe_app #67: Stuck on "waiting for behaviour"

tim-fan, 2020-09-02:
Hi there, I'm trying to run through the basic tutorials, but am not able to get a behaviour to run. It appears to be stuck waiting for confirmation from onboard behaviour.

Not sure if this is a bug or if I've missed something in the setup.

I'm using bionic/melodic

How I set it up:
```sudo apt install ros-melodic-flexbe-behavior-engine
mkdir -p catkin_ws/src
cd catkin_ws/src
git clone https://github.com/FlexBE/flexbe_app.git
cd ..
rosdep install --from-paths src/ --ignore-src
source /opt/ros/melodic/setup.bash
catkin build
source devel/setup.bash
cd src
rosrun flexbe_widget create_repo test_repo
cd ..
catkin build
source devel/setup.bash
roslaunch fexbe_app flexbe_full.launch
```
When the app opens I select "Load Behaviour" then "Example Behaviour". Then I select "Runtime Control" and click "Start Execution". The app shows the screen from the following screenshot, and never transitions past this state ('waiting for confirmation from onboard behaviour'). I waited up to two minutes, but still the screen did not update.
[image]

The full terminal output from roslaunch is:
```$  roslaunch fexbe_app flexbe_full.launch 
... logging to /home/tim/.ros/log/d625547c-ecff-11ea-aecb-3052cb149165/roslaunch-tim-Aspire-R5-471T-32492.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tim-Aspire-R5-471T:35225/

SUMMARY
========

PARAMETERS
 * /behavior/log_enabled: False
 * /behavior/log_folder: ~/.flexbe_logs
 * /behavior/log_level: INFO
 * /behavior/log_serialize: yaml
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    behavior (flexbe_onboard/start_behavior)
    behavior_launcher (flexbe_widget/be_launcher)
    behavior_mirror (flexbe_mirror/behavior_mirror_sm)
    flexbe_app (flexbe_app/run_app)

auto-starting new master
process[master]: started with pid [32502]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d625547c-ecff-11ea-aecb-3052cb149165
process[rosout-1]: started with pid [32514]
started core service [/rosout]
process[behavior_mirror-2]: started with pid [32518]
process[flexbe_app-3]: started with pid [32522]
process[behavior_launcher-4]: started with pid [32524]
process[behavior-5]: started with pid [32525]
[32526:32526:0902/213658.646153:ERROR:component_loader.cc(165)] Failed to parse extension manifest.
[32549:32549:0902/213659.178205:ERROR:sandbox_linux.cc(371)] InitializeSandbox() called with multiple threads in process gpu-process.
[32549:32549:0902/213700.348295:ERROR:buffer_manager.cc(488)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error from previous GL command
(node:32566) [DEP0025] DeprecationWarning: sys is deprecated. Use util instead.
[INFO] [1599039436.337382]: 1 behaviors available, ready for start request.
[INFO] [1599039437.014049]: --- Behavior Engine ready! ---

```

Any ideas on how to get this behaviour to execute?

sashi2295, 2020-09-11:
I have the same issue.

cjue, 2020-09-11:
Can the two of you check which version of the flexbe_onboard/widget packages you are using?

Running the behavior from the create_repo command works fine for me on 16.04 with the current _develop_ branches.

tim-fan, 2020-09-12:
I got those packages from apt (following instructions from http://philserver.bplaced.net/fbe/download.php)

This gives me the following versions:
```
$ rosversion flexbe_widget
1.2.5
$ rosversion flexbe_onboard
1.2.5
```

tim-fan, 2020-09-12:
Follow up, I've confirmed if I build everything from source then the state machine runs (does not get stuck waiting for behaviour).

To be specific, I cloned flexbe_behavior_engine into my source space:
`git clone https://github.com/team-vigir/flexbe_behavior_engine.git`
Then rebuilt, then it all worked.

So it looks like the issue is that the released packages for melodic are not compatible with the master branch of https://github.com/FlexBE/flexbe_app.git

dcconner, 2023-07-24:
Closing this based on lack of recent comments.  If you see issue on consistent Noetic or ros2-devel/humble branches, then please re-open or file a new issue

dcconner, 2023-12-29:
Have you tried pressing the "Connect" button on the Runtime control screen.

Also, verify that all of the ROS nodes are running: the mirror,  be_launcher widget, and onboard are still running.