FlexBE/flexbe_app #68: Can I run flexbe_app at remote PC and connect ROS master on host?

MomoHsieh, 2020-11-06:
Hi, I am a new user of flex_app. It's a great tool to develop ROS and save time!
I want to ask if I just want to run ROS master on the host PC, then run flexbe_app on the remote PC, how can I do that?

Here is my env
host: Ubuntu 18.04 run with  ROS Kinect version.
remote: Win10/Ubuntu18.04 run with flex_app which build by nwjs builder. No ROS system in here.

I can open flex_app normally, but I can't connect to the ROS host. I don't know how to set the ROS URL or what the usage of the flexbe_app the ROS node is wrong. Should I run flex_app on the host PC? Please tell me if this feature can do this.

pschillinger, 2020-11-09:
You can run the flexbe_app on a different PC, most easily using the [standard ROS network setup](http://wiki.ros.org/ROS/NetworkSetup). Similarly, a multi-master solution is possible for more remotely connected machines and FlexBE generally supports this by keeping the required communication bandwidth low.

In all cases, however, it is required that ROS is installed on both machines and also the repository of states and behaviors should be cloned on both machines (for flexbe_app to show and edit, for the robot to execute it). Runtime changes are transferred automatically, but the same base version should be there.

Indeed, many aspects of the flexbe_app also work in Windows. But there is no complete support for it, in particular due to ROS. So if it is a requirement for you to have Windows or at least no ROS on the remote PC, I don't think it will work.

MomoHsieh, 2020-11-12:
@pschillinger Thanks for your reply!

closed 2023-04-28 by dcconner