FlexBE/flexbe_app #72: Is it on the roadmap to make FlexBE App a web service?

tremblay0934, 2021-02-13:
As the question suggest, I would like to know if you have any intention of porting FlexBE App as a deployable web service. This would mean that the web server has acces to the file system of the machine on which it is deployed. A front end web framework would allow to make the graphical state machines. This would be useful because it would allow to :

- Deploy on ARM platforms, like NVIDIA's Jetsons (on which NWJS is not supporter)
- Remote code directly on your robot
- Create easily deployable docker with your robot environment make development accessible to further team members. This is useful because robot prototypes are expansive and not available in numbers equivalent to the dev team members.

If this is not on your roadmap, i would like to know if you have knowledge of anyone who attempted this. If not then what web framework would you suggest that may require less drastic changes to the current code for the desktop FlexBE App.

I am totally aware my requests imply HUGE reworks, so I ask those questions lightly without much hope.
As always, thanks a lot for your great work!

pschillinger, 2021-09-11:
I think there are several nice aspects of having the App as web service, including those that you listed. At the moment, however, it is indeed not on my roadmap due to the following reasons:
* It clearly is a lot of work and therefore, a question of priorities. Even though it has its advantages, I would assume that most people would rather continue to use the App.
* A web browser is a significantly less predictable environment than the App, requiring a lot of additional usability and portability extensions to the current user interface which I cannot fully estimate at the moment.

Still, I'm open for contributions on this topic. Put aside my portability concern, I guess code that need to change or functions to be re-implemented mainly concern code in the `ros` and `io` folders (although only a small subset of `io`). A regular python webserver could then be running as ROS node and provide a websocket connection to forwards ROS messages and other commands that require local execution.

samuk, 2021-10-21:
I'm also interested in this. I wonder if any of the work done on https://github.com/dheera/rosboard is useful?

dcconner, 2023-04-28:
I don't expect movement on this for ROS 1, but we are looking into changes for ROS 2 versions of FlexBE UI going forward.  I invite more discussion on this topic

samuk, 2023-05-18:
Did you have a look at https://github.com/dheera/rosboard#rosboard