FlexBE/flexbe_app #73: Behavior "Polymorphism" almost working but behavior list points to same package.

romayiii, 2021-02-22:
I have different behavior ROS packages for different robot types like this:

- robotA_flexbe_behaviors
- robotB_flexbe_behaviors
- robotC_flexbe_behaviors

in each of this packages there is a behavior that has the same name e.g. "Docking Behavior". This behavior is implemented in different ways in each of the packages. It has the same name, because our user interface doesn't care about the type of robot, but only cares about the task to be executed.

In each of the robots, only the package corresponding to that robot is installed, that means, there is only one behavior called "Docking Behavior" per robot. 

Until this point all seems to work on the onboard side of things. 

However, in the FlexBe App, even though the "Select Behavior" list shows the same behavior three times, when trying to open them, all of the three behaviors point to the same package

The "Example Behavior" created by the "create_repo" script is actually a good way to reproduce this. After creating two new repos, the "Select Behavior" list will show it two times, but clicking on any of them will always open the same behavior from the same package, like this:

1. Example Behavior -> opens the one from robotA_flexbe_behaviors
2. Example Behavior -> opens the one from robotA_flexbe_behaviors

But I would expect it to be working like this:

1. Example Behavior -> opens the one from robotA_flexbe_behaviors
2. Example Behavior -> opens the one from robotB_flexbe_behaviors

Do you have any comments/suggestions/ or hopefully fixes =) on this?

ROS melodic
Flexbe App Version 2.3.0

dcconner, 2023-04-28:
Curious the status of this and related tickets? Did you find a work around?  I don't expect to spend time on this with ROS 1, but will look into this for the ROS 2 versions later this summer

dcconner, 2023-06-29:
There is a new version of ROS 2 that does not address this issue, but does "dump" the list of behaviors and their "key" (changed name to avoid confusion with "behavior_id" being used for check sum.  I need to modify the way the keys are being defined.  The issue is not fixed yet, but it is fixable.

dcconner, 2023-06-29:
This issue is now fixed on the latest ros2-devel branches of flexbe_app and flexbe_behavior_engine as of 30-June-23. 

The solution was to pre-pend the behavior name with "package/".   For example, "test_flexbe_behaviors/Example Behavior".  

Works for behavior launcher or new flexbe app.    Launcher also accepts just the name as old style, but will return some behavior with same name from the first package it encounters when processing the behavior library dictionary.

Note, the first slash delineates the package, so if your name includes a slash (e.g., "Hello World/Universe") then you MUST use the package name with launcher.

romayiii, 2023-06-29:
Awesome, @dcconner! Thanks for the fix. We will definitely use it once we move to ROS2!

dcconner, 2023-06-29:
@romayiii It's a straightforward set of changes if you want to submit a similar PR for ROS 1 on the noetic branch!
[flexbe_app 4004a2a](https://github.com/FlexBE/flexbe_app/commit/4004a2a73ea16525b500d62a55c08dc5a587c172)
[flexbe_behavior_engine a63ef26](https://github.com/FlexBE/flexbe_behavior_engine/commit/a63ef2644cd8afb8f15b163fa7ba5d55895d1986)

closed 2023-07-24 by dcconner