FlexBE/flexbe_app #75: Installed behavior packages in workspace

GabrielHuettenberger, 2021-04-26:
We noticed a problem when working with behavior packages which are both installed and available in the workspace. The expected behavior for these packages would be that the version of the package in the workspace has a higher priority and should be used.

What happens is that for certain FlexBE functionality like opening the behavior code from the app directly the workspace file will be opened but on the other hand it is not possible to save any changes made in the behavior through the FlexBE app since we get the error that the behavior is read only which points to FlexBE trying to use the installed package.
It is also not possible to create new behaviors in a workspace package when the package is installed and we get the same error message.
[image]

dcconner, 2023-04-28:
Is PR #76 intended to address this issue ?

dcconner, 2023-07-24:
See also #81

closed 2023-07-24 by dcconner