FlexBE/flexbe_app #81: can't load the new state package in flexbe

rahulrobots, 2022-09-14:
Hello everyone, 

i created a new state package, but can't see/load that in flexbe. i tried 'force Discover', but that didn't help. 
any help will be highly appreciated. 

Thanks in advance

ingjae, 2022-11-29:
Hello , I just had that problem too
If you are using ubuntu 20.04 ros1 noetic
try develop branch at flexbe_behavior_engine
https://github.com/team-vigir/flexbe_behavior_engine/tree/develop

dcconner, 2023-04-28:
Presume above comment fixed this.  If not, need more information about your set up including OS version, ROS version, and branches for flexbe_behavior_engine and flexbe_app.

I plan to close this issue shortly if I don't hear otherwise.

AravindaDP, 2023-05-05:
Behaviors already exist in source form under behavior packages in ros2 galactic are not discovered by flexbe_app.

Furthermore, newly created behaviors are saved under build and or install directory and not in source directory.

OS: Ubuntu 20.04
ROS version: ros2 galactic
flexbe_behavior_engine branch: ros2-devel
flexbe_app branch: ros2-devel

dcconner, 2023-05-17:
Yes, created behaviors are saved under build/install and must be manually copied to src for saving and distribution.  
If you are defining new states in src, then you first need to make sure they are being installed properly in build/share with colcon before attempting to use them.

New source files need to copied using standard colcon for ROS 2, but we do not yet have a script in place for moving new behaviors that are saved in build over to the "src" folder for saving (e.g. under git repo).  Creating such a script is on our todo list this summer.  it will probably start as a regular script that you run manually, but then integrate into the UI with option to save this way.

dcconner, 2024-01-04:
If you are building from source, use the Noetic branch for both onboard and flexbe_app.
The standard versions of both should now be available as debian installs.

check console messages carefully.  There is a post build step that is required to install the Node JS libraries.  If you try to launch everything in one terminal you might miss the message about running the "nwjs_install" script.

If you have further issues, open a new issue and include some screen shots of your errors.

PanithanRit, 2024-01-17:
Hi deconner

Thank you for your advice!! I did use the Noetic branch and it worked well. However, by the time came I noticed the problem that the app couldn't connect onboard behavior (I saw that other people have the same problem as well on the flexBe bug report). I decided to delete both the flexBe app and onboard package and download again with the Noetic branch (same branch) this time it faced an error like this.
[cid:5d8a6f7f-712e-48f5-b27a-dbf60a744b28]

I assume that because I updated ubuntu maybe there is a problem with the version or something. If you find any solution, please inform me.
Thank you. I still use ubuntu 20.04 LST with Noetic version of ROS

Best
Panithan Rithburi

________________________________
จาก: dcconner ***@***.***>
ส่ง: 5 มกราคม 2567 0:05
ถึง: FlexBE/flexbe_app ***@***.***>
สำเนาถึง: Panithan Rithburi ***@***.***>; Comment ***@***.***>
ชื่อเรื่อง: Re: [FlexBE/flexbe_app] can't load the new state package in flexbe (Issue #81)

If you are building from source, use the Noetic branch for both onboard and flexbe_app.
The standard versions of both should now be available as debian installs.

check console messages carefully. There is a post build step that is required to install the Node JS libraries. If you try to launch everything in one terminal you might miss the message about running the "nwjs_install" script.

If you have further issues, open a new issue and include some screen shots of your errors.

dcconner, 2024-01-17:
I'm not sure what the error code [cid:5d8a6f7f-712e-48f5-b27a-dbf60a744b28] means.

See you posted another comment in #53, possibly the same error.  I'll respond there.  If not, I'll need more information about the error referenced here.