FlexBE/flexbe_app #83: Problem with managing Behaviours in FlexBE

Louismys, 2023-04-03:
Hi!

I'm a relatively new developer and, lately, not long ago, about a week, I started using FlexBE. Now, I'm in a phase of familiarisation with it.
I started using FlexBE App, but I seem to have some problems. Specifically, one problem I face is that I can't manage to delete some behaviours I want to. I can' seem to find any option available to do so and I, really, don't know what to do, especially since I feel that these behaviours, not rightly defined, might cause problems with my use of the App. The problems I'm referring to is that I can't execute the State Machines I created and I, repeatedly, get an error message to which I might refer in the future, if this doesn't get solved by deleting the behaviours I want to.

Can anybody help?

Thank you!

OS: Ubuntu 18.04.1 LTS (Virtual Machine)
ROS Version: Ros Melodic
FlexBE Version: FlexBE 2.3.0

Louismys, 2023-04-03:
Let me report my other issue too:

`
[behavior-4] restarting process
process[behavior-4]: started with pid [5127]
Traceback (most recent call last):
  File "/home/eliasmys/Desktop/ros_workspace/catkin_ws/src/flexbe_behavior_engine/flexbe_onboard/bin/start_behavior", line 18, in <module>
    FlexbeOnboard()
  File "/home/eliasmys/Desktop/ros_workspace/catkin_ws/src/flexbe_behavior_engine/flexbe_onboard/src/flexbe_onboard/flexbe_onboard.py", line 35, in __init__
    self._behavior_lib = BehaviorLibrary()
  File "/home/eliasmys/Desktop/ros_workspace/catkin_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 18, in __init__
    self.parse_packages()
  File "/home/eliasmys/Desktop/ros_workspace/catkin_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 28, in parse_packages
    self._add_behavior_manifests(self._rp.get_path(pkg), pkg)
  File "/home/eliasmys/Desktop/ros_workspace/catkin_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 44, in _add_behavior_manifests
    self._add_behavior_manifests(entry_path, pkg)
  File "/home/eliasmys/Desktop/ros_workspace/catkin_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/behavior_library.py", line 46, in _add_behavior_manifests
    m = ET.parse(entry_path).getroot()
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1182, in parse
    tree.parse(source, parser)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 656, in parse
    parser.feed(data)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1659, in feed
    self._raiseerror(v)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1523, in _raiseerror
    raise err
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 3, column 17
[behavior-4] process has died [pid 5127, exit code 1, cmd /home/eliasmys/Desktop/ros_workspace/catkin_ws/src/flexbe_behavior_engine/flexbe_onboard/bin/start_behavior __name:=behavior __log:=/home/eliasmys/.ros/log/901e263a-d1f8-11ed-ba95-080027a35c7b/behavior-4.log].
log file: /home/eliasmys/.ros/log/901e263a-d1f8-11ed-ba95-080027a35c7b/behavior-4*.log
`
.

That's the error I get. I tried to do something with that, but I couldn't, really, understand.
And the result of the error, I suppose, is the following ('cause every time I try to connect the Runtime it disconnects, after some seconds, while this didn't happen previously):

[image]

.

Any help, especially resolving ones, would be appreciated!

Louismys, 2023-04-03:
(As for the first report, I managed to find a work-around, by deleting the corresponding .py and .xml file for the Behaviours I had in mind. I, just, wanted to report that too. Though, the problem still exists and so does the need for a solution to it. Please, do not avoid it. :) )

dcconner, 2023-04-28:
You are correct in the need to delete both the .py and xml for behaviors. The issue is complicated by how things are managed in src and install/devel folders and there likely interaction with Git, so I don't think we will likely be including any set up for managing behaviors in the ROS 1 version.

As I understand your comments, your second issue disappeared after properly deleting the py and xml files.  

Unless I hear otherwise, I will plan to close this issue shortly.

Louismys, 2023-05-02:
OK, thanks for your answer. You can, now, close the issue. If I have to ask or mention anything else, I may open another issue.