FlexBE/flexbe_app #90: [Noetic] flexbe_app fails to build: missing libraries

sloretz, 2023-07-19:
The ROS Noetic release is failing to build on all platforms. It complains of missing libraries. Maybe there are dependencies not specified in the package.xml? #70 seems like it might be relevant.

https://build.ros.org/job/Nbin_uF64__flexbe_app__ubuntu_focal_amd64__binary/59/consoleFull

```
23:03:15 	dpkg-shlibdeps -Tdebian/ros-noetic-flexbe-app.substvars -l/tmp/binarydeb/ros-noetic-flexbe-app-2.4.1/debian/ros-noetic-flexbe-app//opt/ros/noetic/lib/ debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnode.so debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libffmpeg.so debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/nw debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/swiftshader/libGLESv2.so debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/swiftshader/libEGL.so
23:03:15 dpkg-shlibdeps: warning: cannot find library libnw.so needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnode.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:19 dpkg-shlibdeps: warning: can't extract name and version from library name 'libnw.so'
23:03:19 dpkg-shlibdeps: warning: can't extract name and version from library name 'libnw.so'
23:03:22 dpkg-shlibdeps: warning: cannot find library libffmpeg.so needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: warning: cannot find library libnss3.so needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: warning: cannot find library libnssutil3.so needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: warning: cannot find library libsmime3.so needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: warning: cannot find library libnspr4.so needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libX11-xcb.so.1 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXcomposite.so.1 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXcursor.so.1 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXdamage.so.1 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXfixes.so.3 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXi.so.6 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXrender.so.1 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXtst.so.6 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libcups.so.2 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libdbus-1.so.3 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXss.so.1 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libXrandr.so.2 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libasound.so.2 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libatk-1.0.so.0 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libpangocairo-1.0.so.0 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libpango-1.0.so.0 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libcairo.so.2 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libatspi.so.0 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libgtk-3.so.0 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libgdk-3.so.0 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:22 dpkg-shlibdeps: error: cannot find library libgdk_pixbuf-2.0.so.0 needed by debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
23:03:23 dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnode.so was not linked against libatomic.so.1 (it uses none of the library's symbols)
23:03:23 dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libffmpeg.so debian/ros-noetic-flexbe-app/opt/ros/noetic/share/flexbe_app/nwjs/lib/libnw.so were not linked against librt.so.1 (they use none of the library's symbols)
23:03:23 dpkg-shlibdeps: error: cannot continue due to the errors listed above
23:03:23 Note: libraries are not searched in other binary packages that do not have any shlibs or symbols file.
```

dcconner, 2023-07-19:
Noetic had an old address for maintainer; I've updated and included myself in noetic (ROS 2 was already updated.)

dcconner, 2023-07-19:
@sloretz It appears that the nwjs folder is getting installed as binaries within Debian package.  Normally we run an install script (`rosrun flexbe_app nwjs_install`)  after building to download the relevant nwjs and install.  The CI tests that we run do it this way.  That will clearly have issues if install points to /opt/ros folder, so installing those libraries as part of binary makes sense, but I suspect that some nwjs dependency is not getting captured and installed.  I'll dig into this

dcconner, 2023-07-19:
@pschillinger Any thoughts on this as you handled the earlier (kinetic and before) releases ?

pschillinger, 2023-07-23:
The binary jobs for `flexbe_app` also didn't properly run before, so we always had them excluded. Unfortunately, I don't have a lot of experience with this, but this old PR is where I left it after initially trying to release it: ros/rosdistro#19823. I hope this helps!

dcconner, 2023-07-24:
Thanks.   I missed  that old issue.  I will plan to keep flexbe_app out of releases for now.  I'll revisit this later in fall after we  work on the flexbe_webui more.  I'll leave this open for now to document the issue.