FlexBE/flexbe_app #93: Onboard Status: disconnected

pbrandstet, 2023-10-18:
Hy.
When i run "roslaunch flexbe_app flexbe_full.launch" i always get status disconnected.

I use Parallels Desktop and run Ubuntu 20.04 and ros noetic.
I tried it on a device with Ubuntu 20.04 as native OS too, there it is fine.
In general i never had a problem running Ubuntu or ROS in Parallels Desktop.

On the device with Parallels i also see a toastr with the info "new release available!" I don't see this message on the other device.
Of course i use the noetic branches of the app and the behavior engine.
version: 2.3.0

In the log i see
[INFO] [1697658489.966106]: 1 behaviors available, ready for start request.
[INFO] [1697658490.311659]: --- Behavior Engine ready! ---
[INFO] [1697658490.312348]: No onboard behavior is active.
[INFO] [1697658490.313697]: --- Behavior Mirror ready! ---

i also get some gpu error's, but i dont think that they affect the communication between the behavior engine and the app
[77199:77199:1018/214808.567914:ERROR:component_loader.cc(165)] Failed to parse extension manifest.
[77230:77230:1018/214808.794703:ERROR:sandbox_linux.cc(371)] InitializeSandbox() called with multiple threads in process gpu-process.
[77230:77230:1018/214809.017959:ERROR:buffer_manager.cc(488)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error from previous GL command
[77230:77230:1018/214809.113050:ERROR:gl_surface_presentation_helper.cc(259)] GetVSyncParametersIfAvailable() failed for 1 times!
(node:77252) [DEP0025] DeprecationWarning: sys is deprecated. Use util instead.

Thanks in advance.

zcx0602, 2023-12-27:
I have the same problem, is there a method to solve it?

dcconner, 2024-04-08:
The "new release available" likely means the ROS 2 version, so you can ignore that on FlexBE app.
Be sure you're using the noetic branches, and you must run the nwjs_install script to install the required NWJS binaries prior to launching.