FlexBE/flexbe_app #94: Can't install flexbe_app on ubuntu 20.04 ROS version Noetic

PanithanRit, 2024-01-17:
I have used flexbe since the beginning of Jan 2024. I decided to reinstall flexbe yesterday and I found this problem.
[image]

I use ubuntu 20.04 LTS ROS version : Noetic

I assume that it because of ubuntu update because other packages is working fine. 

Please help

PanithanRit, 2024-01-17:
I try to run this seperately it can't find flexbe_app because it isn't complied yet 
[image]

PanithanRit, 2024-01-17:
I assume that the problem come from this line of code of file nwjs_install 
`curl -s -o $TEMPFILE https://dl.nwjs.io/$VERSION/nwjs-$VERSION-$OS-$ARCH.tar.gz`

When I access website https://dl.nwjs.io and click version " 0.42.3". I got 504 Gateway Time-out error.  Can we replace this file nwjs.tar.gz manually ? or what do I need to add to use flexBe_app.

tcypn, 2024-01-17:
susu na

simonbogh, 2024-02-14:
@PanithanRit did you manage to fix this issue? I have the same problem with a Linux arm install.

dcconner, 2024-04-29:
It's been a while since I tested the noetic version and have never tested an install on Arm.  Did you find a workaround?  Our semester is coming to close, and I'll be working with FlexBE actively this summer (mostly in ROS 2).