FlexBE/flexbe_app #99: FlexBE App saves Behaviors in ./install directory

YBachmann, 2024-03-27:
I noticed that the flexbe app saved my behaviors in the `./install` directory (`./install/foo_state_controller_flexbe_behaviors/local/lib/python3.10/dist-packages/foo_state_controller_flexbe_behaviors`) instead of the my package directory in the `src` folder. 

If I close the flexbe app and launch it again it is able to find the behaviors in the `./install` directory, I can click on "Load Behavior" and load them. But I would really like the flexbe app to store my behaviors directly in my package because otherwise source control / Git gets really messy! 

Here are the contents of `foo_bar01_ws/install/foo_state_controller_flexbe_behaviors`: 
(I changed some names to foo/bar for confidentiality reasons)
``` bash
yannic@yannic-ubuntu-NB002-23:~/projects/foo_GIT/foo_bar01_ws/install/foo_state_controller_flexbe_behaviors$ tree
.
├── lib
│   └── foo_state_controller_flexbe_behaviors
│       ├── config
│       │   └── example.yaml
│       ├── copy_behavior
│       └── manifest
│           ├── example_action_state_behavior.xml
│           ├── example_behavior.xml
│           ├── bar_main_behavior.xml # <--------------------------- My behavior 
│           └── bar_measurement_behavior.xml # <--------------------------- My behavior 
├── local
│   └── lib
│       └── python3.10
│           └── dist-packages
│               ├── foo_state_controller_flexbe_behaviors
│               │   ├── example_action_state_behavior_sm.py
│               │   ├── example_behavior_sm.py
│               │   ├── __init__.py
│               │   ├── bar_main_behavior_sm.py # <--------------------------- My behavior 
│               │   ├── bar_measurement_behavior_sm.py # <--------------------------- My behavior 
│               │   └── __pycache__
│               │       ├── example_action_state_behavior_sm.cpython-310.pyc
│               │       ├── example_behavior_sm.cpython-310.pyc
│               │       ├── __init__.cpython-310.pyc
│               │       └── bar_main_behavior_sm.cpython-310.pyc
│               └── foo_state_controller_flexbe_behaviors-0.0.1-py3.10.egg-info
│                   ├── dependency_links.txt
│                   ├── PKG-INFO
│                   ├── SOURCES.txt
│                   └── top_level.txt
└── share
    ├── ament_index
    │   └── resource_index
    │       ├── package_run_dependencies
    │       │   └── foo_state_controller_flexbe_behaviors
    │       ├── packages
    │       │   └── foo_state_controller_flexbe_behaviors
    │       └── parent_prefix_path
    │           └── foo_state_controller_flexbe_behaviors
    ├── colcon-core
    │   └── packages
    │       └── foo_state_controller_flexbe_behaviors
    └── foo_state_controller_flexbe_behaviors
        ├── cmake
        │   ├── foo_state_controller_flexbe_behaviorsConfig.cmake
        │   └── foo_state_controller_flexbe_behaviorsConfig-version.cmake
        ├── environment
        │   ├── ament_prefix_path.dsv
        │   ├── ament_prefix_path.sh
        │   ├── path.dsv
        │   ├── path.sh
        │   ├── pythonpath.dsv
        │   └── pythonpath.sh
        ├── hook
        │   ├── cmake_prefix_path.dsv
        │   ├── cmake_prefix_path.ps1
        │   └── cmake_prefix_path.sh
        ├── local_setup.bash
        ├── local_setup.dsv
        ├── local_setup.sh
        ├── local_setup.zsh
        ├── package.bash
        ├── package.dsv
        ├── package.ps1
        ├── package.sh
        ├── package.xml
        └── package.zsh

23 directories, 44 files
```

Here are the contents of my package:
``` bash
yannic@yannic-ubuntu-NB002-23:~/projects/foo_GIT/foo_bar01_ws/src/foo_state_controller_behaviors$ tree
.
├── CONTRIBUTING.md
├── LICENSE
├── foo_state_controller_flexbe_behaviors
│   ├── bin
│   │   └── copy_behavior
│   ├── CHANGELOG.rst
│   ├── CMakeLists.txt
│   ├── config
│   │   └── example.yaml
│   ├── foo_state_controller_flexbe_behaviors  # <--------------------------- Only the automatically generated examples 
│   │   ├── example_action_state_behavior_sm.py
│   │   ├── example_behavior_sm.py
│   │   ├── __init__.py
│   ├── manifest   # <--------------------------- Only the automatically generated examples
│   │   ├── example_action_state_behavior.xml
│   │   ├── example_behavior.xml
│   ├── package.xml
│   ├── resource
│   │   └── foo_state_controller_flexbe_behaviors
│   ├── setup.cfg
│   └── setup.py
├── foo_state_controller_flexbe_states
│   ├── CHANGELOG.rst
│   ├── foo_state_controller_flexbe_states
│   │   ├── example_action_state.py
│   │   ├── example_state.py
│   │   ├── __init__.py
│   │   └── foo_bar_baz_state.py
│   ├── package.xml
│   ├── resource
│   │   └── foo_state_controller_flexbe_states
│   ├── setup.cfg
│   ├── setup.py
│   └── tests
│       ├── example_action_state.test
│       ├── example_state.test
│       ├── run_colcon_test.py
│       └── run_tests.launch.py
├── PROJECT_behaviors
│   ├── CHANGELOG.rst
│   ├── CMakeLists.txt
│   └── package.xml
└── README.md

11 directories, 34 files
```

YBachmann, 2024-03-28:
Okay, so I found out that when creating a Behavior Repository with `ros2 run flexbe_widget create_repo my_new_repo` a `copy_behavior` script is automatically generated in `my_new_repo_behaviors/my_new_repoflexbe_behaviors/bin`. 
After setting the environment variable `WORKSPACE_ROOT` to the workspace root directory it is possible to run the `copy_behavior` script for each behavior to copy it from the install directory into the proper place in the package/src-folder. 

> :warning: 
> **This process is neither described in the [FlexBE Tutorials on wiki.ros.org](http://wiki.ros.org/flexbe/Tutorials/Using%20the%20Statemachine%20Editor) nor in the README.md of the [FlexBE App Repo](https://github.com/FlexBE/flexbe_app/tree/humble)!**

Also this is quite tedious and one has to always remember to run the scripts before commiting and pushing changes in the behaviors. 

It would be nice, if the process of copying the behaviors was automatically handeled by FlexBE (maybe even without having to set an environment variable).

YBachmann, 2024-04-02:
@pschillinger I just wanted to kindly ask if I can expect some feedback or maybe even a fix for this issue.

pschillinger, 2024-04-07:
Thanks for pointing out! This is specific to ROS 2 and I think it comes from the difference in how the workspace installation is handled there in comparison to ROS 1. Is it the same behavior when you build the workspace with `--symlink-install`?

Since this is related to ROS 2, @dcconner you might be more familiar with this topic.

YBachmann, 2024-04-08:
It seems that `--symlink-install` did the trick. 
This "workaround" should work for now, however there could be situations where using `--symlink-install` is not preferable/possible due to other packages in the project/workspace.

dcconner, 2024-04-08:
The "copy_behavior" script was the work around for this issue, and you are correct that this is not ideal and has not been updated on tutorials.

We plan work this summer to refresh the documentation for the ROS 2 version.

We are working on a new FlexBE WebUI that will replace the FlexBE App.  This should be released by June 2024.   I propose that we add a configuration option to automatically copy behaviors to src in that version.

dcconner, 2024-06-13:
We've been using the flexbe_webui internally for several weeks.  I have a few minor clean ups to do and then we can push the "beta" release publicly.  It does have the promised functionality to let you optionally save to src folder; you just specify the base of your workspace (e.g ~/ros2_ws/src).  I will endeavor to get that out late today or Friday 14-June.

dcconner, 2024-06-13:
@YBachmann   et al.

FlexBE WebUI is live https://github.com/FlexBE/flexbe_webui .
Right now I'm considering this a beta release and may do some future clean up before an official `ros2-devel` 0.0.0 release.
Please work it hard and let me know if you have any issues.

You may also want to checkout the new documentation at https://flexbe.readthedocs.io/en/latest/

Feedback and comments are appreciated.   Drop me an email at the "contact us" link on documentation, or post issues to the  [FlexBE WebUI issues tab](https://github.com/FlexBE/flexbe_webui/issues) or [Documentation Issues](https://github.com/FlexBE/flexbe_docs/issues).  Both are actively being worked on this summer by my students so get your feedback in early.